ros-industrial-attic/ur_modern_driver #125: Cannot execute trajectories with MoveIt

ipa-mjp, 2017-07-31:
Hello,

I am using UR10 robot arm. I am trying to execute a trajectory with MoveIt and I got the following warning: 

[ WARN] [1501525244.710014591]: Controller arm/joint_trajectory_controller failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1501525244.710173777]: Controller handle arm/joint_trajectory_controller reports status ABORTED.

With this the trajectory is never executed. Any idea what I am setting up wrong?

I tried the same with the normal ur_driver and it works fine.

Thank you so much for the help.

bsantamaria, 2017-08-04:
I'm experiencing the same problem, I'm using ROS Kinetic, and a "Real UR10 with CB2 running 1.8.14035", which is one of the configurations that ThomasTimm tested, so I guess I'm doing something wrong, but I have no idea what it can be...

ipa-mjp, 2017-08-04:
Yes, same things. Our arm version is 1.6 and real UR10 compatible with CB2. I am not sure but I thing so this is moveit error. Might be I am not setting up path constraints in my code and by default it takes zero. Are you set path constraints in your code?

gavanderhoorn, 2017-11-24:
@ipa-mjp: at this point I can't really say what would be wrong.

Have you found out anything in addition to what you reported in your OP?

gavanderhoorn, 2018-03-02:
Closing due to inactivity.

@ipa-mjp: if this is still a problem for you, please re-open and provide more information.

closed 2018-03-02 by gavanderhoorn