ros-industrial-attic/ur_modern_driver #129: There is a solution about catkin_make on ROS kinetic?

Purdiding, 2017-08-23:
Hi, I'm trying to this package on ROS kinetic.
However, I've catkin_make errors.... 

       → ur_modern_driver/CmakeFiles/ur_hardware_interface.dir/src/ur_hardware_interface.cpp.o

Is it support by Kinetic version? How can I do?? 

regards,
Thank you.

eric-breaking, 2017-09-06:
Hi,
i had a similar problem if you change "hardware_interface" with "claimed_resources[0].hardware_interface" in the "ur_hardware_interface.cpp" file, probably it will resolve the problem.

gavanderhoorn, 2017-11-24:
The `indigo-devel` branch does not compile cleanly on Kinetic (or Lunar). This is a known issue at the moment.

If you don't need `ros_control`, you can ignore the error.

If you do, take a look at #120.

gavanderhoorn, 2017-11-24:
I'm closing this one, not to ignore the issue you report, but because we're aware of it and this is essentially a duplicate.

closed 2017-11-24 by gavanderhoorn