ros-industrial-attic/ur_modern_driver #130: Controller  failed with error code INVALID_GOAL

philwall3, 2017-08-24:
Hi,

I'm using the following commands on Ubuntu 14.04 and ROS Indigo:
`roslaunch ur_modern_driver ur5_bringup.launch limited:=true robot_ip:=192.168.20.77`
`roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true`
`roslaunch ur5_moveit_config moveit_rviz.launch config:=true`

The first planning and execution of a move with Rviz works fine and the UR5 moves. But the second time I try to execute a movement, nothing happens.

Here are the 3 terminal outputs:

```
username@pc-5:~$ roslaunch ur_modern_driver ur5_bringup.launch limited:=true robot_ip:=192.168.20.77
... logging to /home/username/.ros/log/b35f4fc4-88c6-11e7-8869-901b0e4e9194/roslaunch-pc-5-901.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /srv-0/username/catkin_ws/src/ur_description/urdf/ur5.urdf.xacro
included from: /srv-0/username/catkin_ws/src/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
started roslaunch server http://192.168.20.105:35708/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/robot_ip_address: 192.168.20.77
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [920]
ROS_MASTER_URI=http://192.168.20.105:11311

setting /run_id to b35f4fc4-88c6-11e7-8869-901b0e4e9194
process[rosout-1]: started with pid [934]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [937]
process[ur_driver-3]: started with pid [947]
[ INFO] [1503577356.012857283]: on_goal
[ INFO] [1503577377.957333687]: on_goal
[ERROR] [1503577377.958174487]: Goal start doesn't match current pose
```

```
username@pc-5:~$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true
... logging to /home/username/.ros/log/b35f4fc4-88c6-11e7-8869-901b0e4e9194/roslaunch-pc-5-1045.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.20.105:45624/

SUMMARY
========

[... 258 lines cut ...]
core service [/rosout] found
process[rviz_pc_5_1493_1465889821850506218-1]: started with pid [1513]
[ INFO] [1503577301.808009809]: rviz version 1.11.15
[ INFO] [1503577301.808063561]: compiled against Qt version 4.8.6
[ INFO] [1503577301.808076239]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1503577302.435783869]: Stereo is NOT SUPPORTED
[ INFO] [1503577302.435880771]: OpenGl version: 4.5 (GLSL 4.5).
[ERROR] [1503577305.934261628]: Skipping XML Document "/opt/ros/indigo/share/industrial_collision_detection/industrial_collision_detection_plugin_description.xml" which had no Root Element.  This likely means the XML is malformed or missing.
[ INFO] [1503577305.939382876]: Loading robot model 'ur5'...
[ INFO] [1503577305.939432841]: No root joint specified. Assuming fixed joint
[ERROR] [1503577305.973997624]: Skipping XML Document "/opt/ros/indigo/share/industrial_collision_detection/industrial_collision_detection_plugin_description.xml" which had no Root Element.  This likely means the XML is malformed or missing.
[ INFO] [1503577306.003565000]: Loading robot model 'ur5'...
[ INFO] [1503577306.003601429]: No root joint specified. Assuming fixed joint
[ INFO] [1503577306.062148732]: Starting scene monitor
[ INFO] [1503577306.065204844]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1503577306.390919820]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1503577307.186025771]: Ready to take MoveGroup commands for group manipulator.
[ INFO] [1503577307.186128492]: Looking around: no
[ INFO] [1503577307.186183628]: Replanning: no
```

With ur5_ros_control.launch instead of ur5_bringup.launch it is working fine after the first movement.
What could be the issue here?

Thanks

gavanderhoorn, 2017-08-24:
from the output of the driver:

> [ERROR] [1503577377.958174487]: Goal start doesn't match current pose

Are you updating the 'current state' in your move group request? The driver compares the first point in the trajectory against the current state. If those don't match, it'll refuse to execute it.

> With ur5_ros_control.launch instead of ur5_bringup.launch it is working fine after the first movement.

Because `ros_control` doesn't implement such a check.

philwall3, 2017-08-26:
Thanks for your answer!
I'm not sure how to do that when only controlling the UR5 through Rviz (very new to this stuff). How and where would I update the 'current state' then?

gavanderhoorn, 2017-08-28:
In most cases that happens automatically with the MoveIt RViz plugin, but you can do that by clicking the `update` button in the `Select Start State` section on the `Planning` tab of the plugin:

[image]

The screenshot shows the `Goal State` section, but the `Start State` section is just behind it.

philwall3, 2017-08-30:
Thanks, updating the 'Start State' after every move to current state works.

What I also found out: You only need to manually update it, when you first only click 'Plan' and then 'Execute' afterwards. When you click 'Plan and Execute' you don't need to manually update the 'Start State'. Is that the normal behavior?

gavanderhoorn, 2017-08-30:
That is MoveIt behaviour, yes.

I believe the rationale is that you can actually set the *start state* to something else than the *current state* (fi to make a plan start somewhere that is not the current state), and if `Plan` would then just update the *start state* automatically, that would not work.

gavanderhoorn, 2017-08-30:
I'm going to close this issue, as it's not really an issue with the driver.

If you feel that is incorrect, please re-open and clarify.

closed 2017-08-30 by gavanderhoorn