ros-industrial-attic/ur_modern_driver #131: Support for the new 5.0.0.0.0 driver

Eirtrand, 2017-08-28:
Im using the new 5.0 driver with ROS. The driver comunicates with the robot, but Im not able to controll it using URScript, test_move.py or moveIt. TF, joint states and everyting else works fine. I get no errors when trying to move the robot, its just stuck on "on_goal" forever.

gavanderhoorn, 2017-08-28:
I'm probably being dense, but what is the `5.0 driver`?

Eirtrand, 2017-08-28:
I meant the URSoftware version 5.0.0.0.0, its a beta software that my ur3 came with. It uses PolyScope 1.0.192.0.0.0 and URControl 1.0.141.0.0.0

gavanderhoorn, 2017-08-28:
Adding support for new system software versions will require publicly available information from Universal Robots.

I suggest you ask UR support to update their documentation on the *client interface* (article 16496).

closed 2017-09-05 by Eirtrand

gavanderhoorn, 2017-09-05:
Could you tell us what the conclusion is / was?

Eirtrand, 2017-09-05:
I didnt have time to look into it, lets hope they make it available soon.

gavanderhoorn, 2018-06-19:
Seems to have taken some time, but `v5` is out (or at least the manuals): [Software_Manual_en_Global.pdf](https://s3-eu-west-1.amazonaws.com/ur-support-site/41109/Software_Manual_en_Global.pdf).