ros-industrial-attic/ur_modern_driver #132: Position controller not working with UR10 and CB2

bsantamaria, 2017-08-30:
I have access to two UR10 robots, one with CB2 and the other with a CB3, I have been trying to use this driver to control them through ros control and I'm experiencing some problems.

With the CB2 I'm able to work using velocity control, it's not very stable but I guess that will be somewhat related to the PID parameters, as I'm using the default ones...
When I try to work with position control I'm facing a bigger issue, the robot moves to the position, but stumbling, I have used a modified ur_hardware_interface.cpp that I have found in another issue to get a graph of the robot movement:

[image]

Does anybody know what could be producing this behaviour?

Thanks in advance

bsantamaria, 2017-08-31:
I have been running some tests and it seems that the CB2 isn't able to work with 0.008 (default) as servoj time parameter, if I increase this value to 0.016 (i.e.), the position control runs as expected. Is this the normal behaviour? If it is, I think it might be a good idea to put a warning somewhere... If there is no one already that I haven't seen...

happygaoxiao, 2017-10-29:
Hello @bsantamaria ，Could you tell me how to using servoj to control the UR robot ? Now I just can use the _ur_driver/joint_speed_ to do trajectory planning. I can't find some examples or tutorials about doing position based ros_control.   Thanks a lot.

bsantamaria, 2017-10-30:
Hello @happygaoxiao , to control the robot with position control using ur_modern_driver you just have to launch ur10_ros_control.launch (or the appropiate version for your robot), and then send trajectories to the specified controller, using moveit! for example.
By default it launches the "vel_based_pos_traj_controller", but you could change this to "pos_pased_pos_traj_controller" by modifying the "ur10_ros_control.launch" file.

willcbaker, 2017-11-16:
more data points: I'm testing on a UR5 CB3 and UR10 CB3 using default ros_control params

The UR10 works nicely in position control, had wild velocities on elbow joint and unsafe
The UR5 Is very jerky in pos control at servoj time 0.008 but seems improved (not perfect) for 0.016
velocity control is OK, again not perfect.

paolovigkiller, 2018-03-02:
Can I Have an example of your test main?