ros-industrial-attic/ur_modern_driver #133: test_move.py => Waiting for server... ||

Eirtrand, 2017-09-06:
ROS Kinetic+Ubuntu 16.04 + MoveIt!

Im starting the simulator like this: 

**roslaunch ur_gazebo ur10.launch limited:=true
roslaunch ur10_moveit_config moveit_rviz.launch config:=true limited:=true 
roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch sim:=true limited:=true**

The planning with rviz and moveit works great, however Im a little skeptic about the "limited" fix for full joint range functionality. For now I have only been using rviz for planning, the problem occur when I started on the moveit tutorials.  The code below is modified from the first example, it gives me the following error:

```bash
[ INFO] [1504685643.025538460]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1504685643.061889690]: Could not find planner plugin name
[ERROR] [1504685643.063331137]: Exception while loading planner '': According to the loaded plugin descriptions the class  with base class type planning_interface::PlannerManager does not exist. Declared types are  chomp_interface/CHOMPPlanner ompl_interface/OMPLPlanner
Available plugins: chomp_interface/CHOMPPlanner ompl_interface/OMPLPlanner 
```
 Also the test_move.py gives me a :-1: by being stuck waiting on server. Does it exist a working example for using path planning with the ur robot and moveit without rviz?  [This](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html#planning-to-a-pose-goal) Is the example Im trying to get working

```cpp

#include <pluginlib/class_loader.h>
#include <ros/ros.h>

#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/planning_interface/planning_interface.h>
#include <moveit/planning_scene/planning_scene.h>
#include <moveit/kinematic_constraints/utils.h>
#include <moveit_msgs/DisplayTrajectory.h>
#include <moveit_msgs/PlanningScene.h>

#include <boost/scoped_ptr.hpp>

int main(int argc, char** argv)
{
  ros::init(argc, argv, "planning");
  ros::AsyncSpinner spinner(1);
  spinner.start();
  ros::NodeHandle node_handle("~");

  
  robot_model_loader::RobotModelLoader robot_model_loader("robot_description");
  robot_model::RobotModelPtr robot_model = robot_model_loader.getModel();

  planning_scene::PlanningScenePtr planning_scene(new planning_scene::PlanningScene(robot_model));

  // We will now construct a loader to load a planner, by name.
  // Note that we are using the ROS pluginlib library here.
  boost::scoped_ptr<pluginlib::ClassLoader<planning_interface::PlannerManager>> planner_plugin_loader;
  planning_interface::PlannerManagerPtr planner_instance;
  std::string planner_plugin_name;

  // We will get the name of planning plugin we want to load
  // from the ROS param server, and then load the planner
  // making sure to catch all exceptions.
  if (!node_handle.getParam("chomp_interface/CHOMPPlanner", planner_plugin_name))
    ROS_FATAL_STREAM("Could not find planner plugin name");
  try
  {
    planner_plugin_loader.reset(new pluginlib::ClassLoader<planning_interface::PlannerManager>(
        "moveit_core", "planning_interface::PlannerManager"));
  }
  catch (pluginlib::PluginlibException& ex)
  {
    ROS_FATAL_STREAM("Exception while creating planning plugin loader " << ex.what());
  }
  try
  {
    planner_instance.reset(planner_plugin_loader->createUnmanagedInstance(planner_plugin_name));
    if (!planner_instance->initialize(robot_model, node_handle.getNamespace()))
      ROS_FATAL_STREAM("Could not initialize planner instance");
[... 153 lines cut ...]
  // We use a bound that definitely includes the reachable space for the arm. This is fine because sampling is not done
  // in this volume
  // when planning for the arm; the bounds are only used to determine if the sampled configurations are valid.
  req.workspace_parameters.min_corner.x = req.workspace_parameters.min_corner.y =
      req.workspace_parameters.min_corner.z = -2.0;
  req.workspace_parameters.max_corner.x = req.workspace_parameters.max_corner.y =
      req.workspace_parameters.max_corner.z = 2.0;

  // Call the planner and visualize all the plans created so far.
  context = planner_instance->getPlanningContext(planning_scene, req, res.error_code_);
  context->solve(res);
  res.getMessage(response);
  display_trajectory.trajectory.push_back(response.trajectory);
  // Now you should see four planned trajectories in series
  display_publisher.publish(display_trajectory);

  // END_TUTORIAL
  sleep_time.sleep();
  ROS_INFO("Done");
  planner_instance.reset();

  return 0;
}

```

Eirtrand, 2017-09-06:
Update: Fixed some obvious problems in the code, the error is now 

```bash
[ INFO] [1504702414.601034944]: Loading robot model 'ur10'...
[ INFO] [1504702414.601133716]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1504702414.685288912]: Loading robot model 'ur10'...
[ INFO] [1504702414.685351356]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1504702414.761625012]: Initializing OMPL interface using ROS parameters
[ INFO] [1504702414.775779668]: Using planning interface 'OMPL'
[ERROR] [1504702429.777157323]: Found empty JointState message
[ INFO] [1504702429.777343748]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[ WARN] [1504702429.777434979]: It looks like the planning volume was not specified.
[ WARN] [1504702429.777670746]: Returning dirty link transforms
[ WARN] [1504702429.777794063]: Returning dirty link transforms
[ WARN] [1504702429.778247850]: Returning dirty link transforms
[ WARN] [1504702429.778305998]: Returning dirty link transforms
[ INFO] [1504702429.779020260]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1504702429.780155280]: RRTConnect: Starting planning with 1 states already in datastructure
[ERROR] [1504702430.789017379]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1504702430.789058742]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1504702430.789078385]: No solution found after 1.009459 seconds
[ INFO] [1504702430.817945384]: Unable to solve the planning problem
[ERROR] [1504702430.818016579]: Could not compute plan successfully
[ WARN] [1504702430.820508344]: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.

```

gavanderhoorn, 2017-09-06:
I can't know for certain without knowing your setup in more detail, but as I wrote on https://github.com/ros-industrial/universal_robot/issues/318, what you're trying to do is absolutely possible.

Note btw: you're using a simulation, this issue tracker is not the correct place to report this, as the driver is not involved at all at this point.

gavanderhoorn, 2017-11-24:
As this seems / seemed to be an issue with MoveIt and / or configuration, not with the package(s) in this repository, I'm going to close this.

gavanderhoorn, 2017-11-24:
@Eirtrand: the MoveIt issues can definitely be solved, but I would advise you to ask for support on ROS Answers or the MoveIt issue tracker (preferably the former).

closed 2017-11-24 by gavanderhoorn