ros-industrial-attic/ur_modern_driver #134: Installation instruction unclear

MathijsNL, 2017-09-09:
Hi there,

> Installation
> 
> Just clone the repository into your catkin working directory and make it with catkin_make.

Should be changed to:

> Installation
> 
> Just clone the repository into your catkin/src directory and make it with catkin_make.

Because right now it gives the impression that it should be in the catckin_ws root.

gavanderhoorn, 2017-11-24:
Thanks for reporting this.

I believe we assume a certain level of familiarity with ROS & Catkin concepts, but we can probably add something like "clone is into the source space of your Catkin workspace".

ipa-nhg, 2019-07-02:
Fixed: https://github.com/ros-industrial/ur_modern_driver/pull/319

gavanderhoorn, 2019-07-02:
Fixed in #319.

closed 2019-07-02 by gavanderhoorn