ros-industrial-attic/ur_modern_driver #135: Problem in instalation ur_modern_driver catkin_make on ROS kinetic!

haniehes, 2017-09-21:
I am trying to make a contact with UR5 trough ROS kinetic which is installed on Ubuntu 16.04. So I tried to clone the ur_modern_driver and I called catkin_make. But it cannot be done and after many errors, it shows `Invoking "make -j4 -l4" failed`. Also my errors start from this: `/home/hanieh/catkin_ws/src/ur_modern_driver/src/ur_hardware_interface.cpp:180:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’`
What is exactly wrong with me.

nano-meter, 2017-09-21:
Apply [these](https://github.com/iron-ox/ur_modern_driver/commit/883070d0b6c0c32b78bb1ca7155b8f3a1ead416c) changes from #96 to the src/ur_hardware_interface.cpp file and it will compile and run, or you can wait for #120, which is a full refactor to be pulled.

haniehes, 2017-09-25:
Thanks for your help. It works well.

v4hn, 2017-10-03:
please close this if your problem is solved.

closed 2017-10-03 by haniehes

ChelseyM, 2018-10-10:
Hey! It appears that while #120 is now merged, it doesn't include the changes from #96, which still need to be applied in order for `catkin_make` to work successfully

gavanderhoorn, 2018-10-10:
Can you make sure you are building the `kinetic-devel` branch of `ros-industrial/ur_modern_driver`?

`master` of that repository is not Kinetic compatible.