ros-industrial-attic/ur_modern_driver #136: ROS Kinetic Ubuntu 16.04 can't move UR5 arm

kzhang8850, 2017-09-29:
Hi All,

Trying to use the updated version of this repository to run my UR5, to no success. It compiles fine, and I can run the launch file, but then get this;
'''
... logging to /home/rooster/.ros/log/02cef73c-a567-11e7-a325-34e6d7875778/roslaunch-bakbak-23511.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /opt/ros/kinetic/share/ur_description/urdf/ur5.urdf.xacro
included from: /opt/ros/kinetic/share/ur_description/urdf/ur5_robot.urdf.xacro
started roslaunch server http://bakbak:41307/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/robot_ip_address: 10.253.0.51
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [23525]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 02cef73c-a567-11e7-a325-34e6d7875778
process[rosout-1]: started with pid [23538]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [23549]
process[ur_driver-3]: started with pid [23552]
[ INFO] [1506724751.253295153]: Setting up connection: 10.253.0.51:30003
[ INFO] [1506724751.253814337]: Connection established for 10.253.0.51:30003
[ INFO] [1506724751.253859833]: Setting up connection: 10.253.0.51:30001
[ INFO] [1506724751.254040080]: Connection established for 10.253.0.51:30001
[ INFO] [1506724751.329372335]: Got VersionMessage:
[ INFO] [1506724751.329400925]: project name: URControl
[ INFO] [1506724751.329415099]: version: 3.2.20175
[ INFO] [1506724751.329426178]: build date: 29 2016, 10:43:26
[ INFO] [1506724751.329451577]: Disconnecting from 10.253.0.51:30001
[ INFO] [1506724751.337893052]: Setting up connection: :50001
[ INFO] [1506724751.338021093]: Connection established for :50001
[ INFO] [1506724751.338052217]: ActionServer enabled
[ INFO] [1506724751.340201355]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1506724751.340242393]: Starting main loop
[ INFO] [1506724751.340343816]: Setting up connection: 10.253.0.51:30003
[ INFO] [1506724751.340656763]: Connection established for 10.253.0.51:30003
[ INFO] [1506724751.340789728]: Setting up connection: 10.253.0.51:30002
[ INFO] [1506724751.342178330]: Connection established for 10.253.0.51:30002
[ INFO] [1506724751.344847191]: Starting ActionServer
[ INFO] [1506724751.345115361]: Trajectory thread started
[ERROR] [1506724751.431210327]: Sub-package of type 0 was not parsed completely!
[ INFO] [1506724751.431262596]: Disconnecting from 10.253.0.51:30002 '''

I just run roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=10.253.0.51 when doing this. I followed this link for steps on how to connect and test move the arm, except just replace some of the steps with kinetic versions/ this repository (even Zagitta's version)/etc.

The UR5 arm's network setup is the same, except its address is 10.253.0.51.

Any ideas on what I'm doing wrong?

gavanderhoorn, 2017-11-24:
> `[ERROR] [..] Sub-package of type 0 was not parsed completely!`

This seems to point to some issue with either deserialisation or a network issue.

>  I followed this link (https://github.com/olinrobotics/irl/wiki/Tutorial:-Setting-up-UR5-robot-arm) for steps on how to connect and test move the arm, except just replace some of the steps with kinetic versions/ this repository (even Zagitta's version)/etc.

Did you also edit the source as instructed in that tutorial? If so, please revert the changes.

gavanderhoorn, 2018-03-02:
Closing due to inactivity.

@kzhang8850: if this is still a problem for you, please re-open and provide more information.

Chunting, 2018-10-05:
Hi Gavanderhoorn,
I met the same problem. Would you please reopen this issue?
When I run the command oslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102,
the error message shows up as follows:
[image]

Thanks,
Chunting

gavanderhoorn, 2018-10-05:
Just for tracking this: can you tell us which version of ROS, which OS (and version), how you installed ROS, how you built `ur_modern_driver`, which robot you have, the version of polyscope it is running, etc.

Also: could you please *not* use screenshots to show me terminal output? Please just copy-paste it into your comment.

Chunting, 2018-10-05:
OK, thanks for your quick reply.
I'm working Ubuntu 16.04, ROS-Kinetic, ur_modern_driver and Polyscope 3.2.21545.
To install ROS, I followed the instructions [here](http://wiki.ros.org/kinetic/Installation/Ubuntu).

As for the error message, it comes like:

> [ INFO] [1538722053.643268977]: Trajectory thread started
**[ERROR] [1538722053.672768737]: Sub-package of type 0 was not parsed completely!**
[ INFO] [1538722053.672891508]: Disconnecting from 192.168.1.102:30002
[ INFO] [1538722053.673016779]: Shutting down on stopped pipeline StatePacket
* Error in `/home/chunting/catkin_ws/devel/lib/ur_modern_driver/ur_driver': double free or corruption (!prev): 0x0000000000fadf10 *
[ur_driver-3] process has died [pid 26071, exit code -11, cmd /home/chunting/catkin_ws/devel/lib/ur_modern_driver/ur_driver __name:=ur_driver __log:=/home/chunting/.ros/log/890f3b82-c86a-11e8-9fd8-44850011896a/ur_driver-3.log].
log file: /home/chunting/.ros/log/890f3b82-c86a-11e8-9fd8-44850011896a/ur_driver-3*.log

It seems that the action server didn't run properly, it stopped in the if statement. 
`void ActionServer::trajectoryThread()
{
  LOG_INFO("Trajectory thread started");

  while (running_)
  {
    LOG_INFO("running_ is true");
    std::unique_lock<std::mutex> lk(tj_mutex_);
    if (!tj_cv_.wait_for(lk, std::chrono::milliseconds(100), [&] { return running_ && has_goal_; })){
        continue;
    }`

Thanks,
Chunting

gavanderhoorn, 2018-10-05:
I don't believe the actionserver is really involved here: parsing the binary stream coming from the controller failed, which causes the whole driver to shutdown (instead of ignoring data and plodding on).

> Polyscope 3.2.21545

it could be there is an incompatibility between that version and the current deserialisers in the driver.

Do you have a specific reason for staying with 3.2.x?

Chunting, 2018-10-05:
No, this version came with the robot and I never updated it.
So probably I have two solutions, one  is to update the polyscope, the other is to go back to Indigo.
Is it?

Thanks,
Chunting

gavanderhoorn, 2018-10-05:
> No, this version came with the robot and I never updated it.
> So probably I have two solutions, one is to update the polyscope,  [..]

It would be interesting to know whether an update of your controller affects the problem, yes.

> the other is to go back to Indigo. Is it?

I'm not sure I understand. Why would downgrading to Indigo help?

AustinDeric, 2018-10-11:
We are running the new e-series UR-10 with firmware version 5.1.0.40195 (Aug 22 2018) and  we are seeing an identical error.

gavanderhoorn, 2018-10-11:
Which would make sense seeing as the binary protocol was again changed but the deserialisers weren't yet.

wanglong06, 2018-12-04:
I was trying to set up a UR5 this morning, and I encountered the exact same problem. The software version was **3.2.18654**

Thanks to @miguelprada and @gavanderhoorn 's advice, I upgraded the software to **3.7.2.40245**, then the problem is solved.

Thanks,
Long.

brianzhang-git, 2018-12-06:
Same error as everyone, on a UR5e. @AustinDeric did you manage to find a workaround? I've been using dniewinski's forks for `universal_robot` and `ur_modern_driver` packages with the appropriate branches but am still running into this error.

Anastasiamor100, 2019-02-07:
Hi, I was experiencing this same issue. I have tried implementing the @AustinDeric solution however my robot does not move when I run the test_move.py file or attempt movement using rviz. 
I am using Ubuntu 18.04, ros melodic and a UR5e robot with ursoftware 5.0.4.x.

I receive the following errors when i use the command: 
`rosrun test_move.py`
```
Waiting for server...
Connected to server
This program makes the robot move between the following three poses:
[126.05071492878113, 0.0, -89.95437383553926, 0.0, 0.0, 0.0]
[85.94366926962348, 0.0, -89.95437383553926, 0.0, 0.0, 0.0]
[85.94366926962348, -11.459155902616464, -89.95437383553926, 0.0, 0.0, 0.0]
Please make sure that your robot can move freely between these poses before proceeding!
Continue? y/n: y
```
AND
```
roslaunch ur_modern_driver  ur5e_bringup.launch robot_ip:=192.168.1.100 [reverse_port:=REVERSE_PORT]
```
```
... logging to /home/blamboozel/.ros/log/3e4a4e28-2a87-11e9-806d-45b75f9e99eb/roslaunch-dreamcatcher-16161.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://dreamcatcher:33101/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/robot_ip_address: 192.168.1.100
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/shutdown_on_disconnect: True
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [16174]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3e4a4e28-2a87-11e9-806d-45b75f9e99eb
process[rosout-1]: started with pid [16185]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [16192]
process[ur_driver-3]: started with pid [16193]
[ INFO] [1549509597.417524971]: Setting up connection: 192.168.1.100:30003
[ INFO] [1549509597.417891165]: Connection established for 192.168.1.100:30003
[ INFO] [1549509597.417922196]: Setting up connection: 192.168.1.100:30001
[ INFO] [1549509597.418118992]: Connection established for 192.168.1.100:30001
[ INFO] [1549509597.418344500]: Got VersionMessage:
[ INFO] [1549509597.418354535]: project name: URControl
[ INFO] [1549509597.418363897]: version: 5.0.0
[ INFO] [1549509597.418383207]: build date: 01-06-2018, 15:28:58
[ INFO] [1549509597.418393085]: Disconnecting from 192.168.1.100:30001
[ INFO] [1549509597.422056687]: ActionServer enabled
[... 70 lines cut ...]
  movej([1.5, 0, -1.57, 0, 0, 0], t=1, r=0)
  movej([1.5, -0.2, -1.57, 0, 0, 0], t=1, r=0)
  movej([2.2, 0, -1.57, 0, 0, 0], t=2, r=0)
  movej([1.5, 0, -1.57, 0, 0, 0], t=1, r=0)
  movej([1.5, -0.2, -1.57, 0, 0, 0], t=1, r=0)
  movej([2.2, 0, -1.57, 0, 0, 0], t=2, r=0)
  movej([1.5, 0, -1.57, 0, 0, 0], t=1, r=0)
  movej([1.5, -0.2, -1.57, 0, 0, 0], t=1, r=0)
  movej([2.2, 0, -1.57, 0, 0, 0], t=2, r=0)
  movej([1.5, 0, -1.57, 0, 0, 0], t=1, r=0)
  movej([1.5, -0.2, -1.57, 0, 0, 0], t=1, r=0)
  movej([2.2, 0, -1.57, 0, 0, 0], t=2, r=0)
  movej([1.5, 0, -1.57, 0, 0, 0], t=1, r=0)
  movej([1.5, -0.2, -1.57, 0, 0, 0], t=1, r=0)
  stopj(3.14)

  socket_close()
end

[ INFO] [1549509786.685703607]: Setting up connection: :50001
[ INFO] [1549509786.685950971]: Connection established for :50001
```

These messages are coming from the trajectory_follower.cpp file. 
Does anyone know why this is happening?

Jason-Zhou1, 2019-03-20:
Hi Anastasiamor100,
I am also using Ubuntu 18.04, ros melodic and a UR5e robot with ursoftware 5.0.2. May I know how you solved the not movable issue at last? Where did you get the "ur5e_bringup.launch" file?

Jason-Zhou1, 2019-03-20:
My current status is:
I get the "ur5e_bringup.launch" file by editing "ur5_bringup.launch" (only changing "$(find ur_description)/launch/ur5_uplaod.launch" into "$(find ur_e_description)/launch/ur5e_uplaod.launch"). 

I launch the ur5e_bringup file and run the test_move.py file in new terminal. When it displays: Continue: y/n. I key in y. Then nothing happens any more. In the ur5e_bringup command line, it displays: on_goal.

Then I run the Moveit and Rviz. The Rviz can update the display of the state of the real ur5e arm, and the "Plan" button also works smoothly, but the "Execute" button does not take effect.

In summary, my robot state is well updated, but I cannot control the movement of the ur5e real arm through the ROS, even though the PolyScope can still drive the arm.

For the IP setting in PolyScope, I keep "Default gateway","Preferred/Alternative DNS Server" as: 0.0.0.0, and I also Enable the Remote Control.

What should I do next to make the arm move through ROS in my laptop?

Jason-Zhou1, 2019-03-20:
My current status is:
 I get the "ur5e_bringup.launch" file by editing "ur5_bringup.launch" (only changing "$(find ur_description)/launch/ur5_uplaod.launch" into "$(find ur_e_description)/launch/ur5e_upload.launch").

I launch the ur5e_bringup file and run the test_move.py file in new terminal. When it displays: Continue: y/n. I key in y. Then nothing happens any more. In the ur5e_bringup command line, it displays: on_goal.

Then I run the Moveit and Rviz. The Rviz can update the display of the state of the real ur5e arm, and the "Plan" button also works smoothly, but the "Execute" button does not take effect.

In summary, my robot state is well updated, but I cannot control the movement of the ur5e real arm through the ROS, even though the PolyScope can still drive the arm.

For the IP setting in PolyScope, I keep "Default gateway","Preferred/Alternative DNS Server" as: 0.0.0.0, and I also Enable the Remote Control.

What should I do next to make the arm move through ROS in my laptop?

Any suggestion is much appreciated.

Jason-Zhou1, 2019-03-20:
Hi Anastasiamor100,
I have found that my command lines do not display the following messages as yours displayed:
...
[INFO]...
[ERROR]...
socket open() ...
[MOVEJ ...
socket close() ...

Does that mean my computer has not setup connection with the real robot arm?

Anastasiamor100, 2019-03-21:
> Hi Anastasiamor100,
> I have found that my command lines do not display the following messages as yours displayed:
> 
> ```
> ...
> [INFO]...
> [ERROR]...
> socket open() ...
> [MOVEJ ...
> socket close() ...
> ```
> Does that mean my computer has not setup connection with the real robot arm?

Hi Jason Zhou1, I'm was using this version of the code, https://github.com/ros-industrial/ur_modern_driver/issues/136#issuecomment-444709653

Also make sure to select the remote mode key next to the hamburger menu mentioned in https://github.com/ros-industrial/ur_modern_driver/issues/184#issuecomment-435325436

Jason-Zhou1, 2019-03-22:
> Hi Jason Zhou1, I'm was using this version of the code, #136 (comment)
> mentioned in #184 (comment)

Hi Anastasiamor100,
Ok, I will try it following your advice.
By the way, do you have idea about how to setup ur5e ip address? I don't
know how to set the gateway, DNS server addresses.
Thank you very much for your help!
Best regards
Jason

Jason-Zhou1, 2019-03-22:
Hi Anastasiamor100,

When I download the 2 packages (the updated PlusOne's universal_robot
<https://github.com/plusone-robotics/universal_robot/tree/ur_e_fix>
and PlusOne's
ur_modern_driver
<https://github.com/plusone-robotics/ur_modern_driver/tree/ur_moderner_driver>)
and catkin_make the catkin_ws, i got error massages:
------------------------------------------------------------------------------------------------------------------
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83
(find_package):
  Could not find a package configuration file provided by "industrial_msgs"
  with any of the following names:

    industrial_msgsConfig.cmake
    industrial_msgs-config.cmake

  Add the installation prefix of "industrial_msgs" to CMAKE_PREFIX_PATH or
  set "industrial_msgs_DIR" to a directory containing one of the above
files.
  If "industrial_msgs" provides a separate development package or SDK, be
  sure it has been installed.
-------------------------------------------------------------------------------------------------------------------------

Then I downloaded industrial_core (which inlcudes industrial_msgs) to my
catkin_ws:
git clone https://github.com/ros-industrial/industrial_core.git
Then i catkin_make and get messages as:
-----------------------------------------
In file included from
/opt/ros/melodic/include/hardware_interface/robot_hw.h:36:0,
                 from
/opt/ros/melodic/include/controller_interface/controller_base.h:37,
                 from
/opt/ros/melodic/include/controller_manager/controller_spec.h:41,
                 from
/opt/ros/melodic/include/controller_manager/controller_manager.h:36,
                 from
/home/longjiang/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:2,
                 from
/home/longjiang/catkin_ws/src/ur_modern_driver/src/ros/hardware_interface.cpp:1:
/opt/ros/melodic/include/hardware_interface/internal/interface_manager.h:69:10:
warning: type qualifiers ignored on function return type
[-Wignored-qualifiers]
   static const void callConcatManagers(typename std::vector<T*>& managers,
T* result)
          ^~~~~
In file included from
/opt/ros/melodic/include/hardware_interface/robot_hw.h:36:0,
                 from
/opt/ros/melodic/include/controller_interface/controller_base.h:37,
                 from
/opt/ros/melodic/include/controller_manager/controller_spec.h:41,
                 from
/opt/ros/melodic/include/controller_manager/controller_manager.h:36,
                 from
/home/longjiang/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ros/controller.h:2,
                 from
/home/longjiang/catkin_ws/src/ur_modern_driver/src/ros/controller.cpp:1:
/opt/ros/melodic/include/hardware_interface/internal/interface_manager.h:69:10:
warning: type qualifiers ignored on function return type
[-Wignored-qualifiers]
   static const void callConcatManagers(typename std::vector<T*>& managers,
T* result)
          ^~~~~
In file included from
/home/longjiang/catkin_ws/src/ur_modern_driver/src/ros/hardware_interface.cpp:1:0:
/home/longjiang/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:
In constructor ‘VelocityInterface::VelocityInterface(URCommander&,
hardware_interface::JointStateInterface&,
std::vector<std::__cxx11::basic_string<char> >&, double)’:
/home/longjiang/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:57:10:
warning: ‘VelocityInterface::max_vel_change_’ will be initialized after
[-Wreorder]
   double max_vel_change_;
          ^~~~~~~~~~~~~~~
/home/longjiang/catkin_ws/src/ur_modern_driver/include/ur_modern_driver/ros/hardware_interface.h:56:40:
warning:   ‘std::array<double, 6> VelocityInterface::prev_velocity_cmd_’
[-Wreorder]
   std::array<double, 6> velocity_cmd_, prev_velocity_cmd_;
                                        ^~~~~~~~~~~~~~~~~~
/home/longjiang/catkin_ws/src/ur_modern_driver/src/ros/hardware_interface.cpp:32:1:
warning:   when initialized here [-Wreorder]
 VelocityInterface::VelocityInterface(URCommander &commander,
hardware_interface::JointStateInterface &js_interface,
 ^~~~~~~~~~~~~~~~~
[  2%] Linking CXX shared library
/home/longjiang/catkin_ws/devel/lib/libur_hardware_interface.so
[  2%] Built target ur_hardware_interface
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
-------------------------------------------------------------------------------------------------------------------------------

Later I used the industrial_msgs instead of industrial_core and the error
messages are still there.

What do you think is the problem in my operations?

Thank you!

Best regards

Jason

miguelprada, 2019-03-22:
@Jason-Zhou1 this issue is supposed to track the problem with deserialization of certain packages for certain PolyScope versions. Please refrain from using it as a general Q&A discussion forum; you can use https://answers.ros.org for that.

I'm going to close the issue since it has long gone off-topic.

closed 2019-03-22 by miguelprada

sourabhacharya, 2019-06-15:
Hello @miguelprada 
Since you have experience on this issue, could you please help me troubleshoot this error?
I'm getting the same error on ROS Kinetic being run on Debian 8 with UR10 CB3 series.
Thanks!
----------------------------------------------------------------------------------------------------------
[ INFO] [1560555640.026144442]: Trajectory thread started                              
[ERROR] [1560555640.092470129]: Sub-package of type 0 was not parsed completely!   
[ INFO] [1560555640.092610956]: Disconnecting from 10.45.93.4:30002                   
[ INFO] [1560555640.092722969]: Shutting down on stopped pipeline StatePacket          
[ur_driver-2] process has died [pid 1824, exit code -11
----------------------------------------------------------------------------------------------------------