ros-industrial-attic/ur_modern_driver #137: UR3 breaking/wobbling problem

htugal, 2017-10-03:
Hi, 
In order to control robot, I publish the ur_driver/joint_speed based on some calculations and the robot subscribes it, yet  robot occasionally breaks for a really short period of time and continues to operate normally. I am not sure whether it actually breaks or not but the smooth movements interrupted like a breaking/wobbling behaviour. Frequency of this breaking behaviour looks random, couldn't find any relationship with anything and running out of options..

I  wonder whether   anybody  has ever come cross such behaviour in UR3 or not ?

happygaoxiao, 2017-10-13:
I have also come cross this behavior. There are some zero value of joint speed as issuse [#139](https://github.com/ThomasTimm/ur_modern_driver/issues/139)

htugal, 2017-10-13:
Hi,
I thought that might be due to package loss/drop within the network communication, yet same behaviours appear even if I use a port switch, here is the first three joints ref and actual velocities: #139 

[image]

gavanderhoorn, 2017-11-24:
Even though #139 was reported later, I believe it is a similar enough issue to suggest we continue trying to diagnose it there, and close this one.

I see you've already commented on #139 as well @htugal.

closed 2017-11-24 by gavanderhoorn