ros-industrial-attic/ur_modern_driver #146: How to get external torques of the UR Robot?

SnoOog, 2017-11-10:
Hello,

I am working with a UR10 Robot and would need the external joint torques (torques caused by external forces) available in ROS for my control structure.

There is the URScript command "get_joint_torques" that give me exactly this values. I managed to write a script running on the robot that sends the returned 6float vector via a socket connection - BUT: as soon as i perform any action or any other command is send to the robot (i.e. a position command via the ur_modern_driver), the script running on the robot stops and does not send any more values.

As far as I know, the ur_modern_driver uses the information, that is published by the UR controller on ports 30002 & 300003, which includes position, vlocity, etc. and only motor currents - no torque nor external torques. 

Is there a way to get the external torques for real-time compliance control? Preferably also streamed with 125 Hz like the other values.

Thanks for your help,
kind Regards,
Matthias

gavanderhoorn, 2017-11-24:
If that data is available through either the secondary (`TCP30002`) or realtime interface (`TCP30003`) we can add publishers for it to the driver.

If not, it's going to be more difficult, as we would have to extent the small bit of URScript that is run by the driver on the controller.

That in itself is not difficult, but the whole point of `ur_modern_driver` was to run as little URScript on the controller itself - and then only when it's really needed.

> As far as I know, the ur_modern_driver uses the information, that is published by the UR controller on ports 30002 & 300003, which includes position, vlocity, etc. and only motor currents - no torque nor external torques.

If you can figure out whether this is true (ie: the data not being available in sec and rt streams) then we can decide on a possible approach.

Thompson104, 2018-01-26:
@SnoOog Perhaps you can send URScript to the topic "pub /ur_driver/URScript"? I haven't tried it myself, but I think it might work.

gavanderhoorn, 2018-01-26:
I'm closing due to inactivity.

If/when someone figures out whether that data is available we can update the protocol parsers, but I believe the ball is in URs court, as joint torques and external forces are not part of the data broadcast on any of the ports we connect to.

gavanderhoorn, 2018-01-26:
I thought that perhaps the RTDE could help us here, but according to the documentation, the data that is accessible through that also doesn't contain this info.

(I seem to remember that RTDE essentially makes use of the same datastream we do in the driver(s), so that doesn't really surprise me)

closed 2018-01-26 by gavanderhoorn