ros-industrial-attic/ur_modern_driver #147: Support for Version 3.4

anjalirao1, 2017-11-10:
I am using v3.43.77 and  I am unable to connect with the robot. I get the following error:
`Error connecting to get firmware version`
 
I am working on Ubuntu 16.04, working with ur_modern_driver. 
I have looked at other posts with the same error message, however there is no conclusion/ solution in these posts.
.
I have disconnected the Wifi, made sure I have sourced my workspace bash script Kindly help

workproduct, 2017-11-19:
also interested in v3.4.3.77 support

naijoug, 2017-11-24:
@anjalirao1  I use `ur_modern_driver` to move UR3/CB3 3.4.0，it is ok . 
Are you using a virtual machine?  
if virtual machine, network setting must be  `bridged adaptor`.

gavanderhoorn, 2017-11-24:
@naijoug: I'm confused: in #150 you report an issue with trying to use `ur_modern_driver` with `3.4.0`. Now you are saying it works.

Which is it?

naijoug, 2017-11-24:
@gavanderhoorn  I use `ur_modern_driver` move UR3 .  If the distance short, move success.
If the distance far , UR3 move a short distance(Not reached the goal)，stop and PolyScope show the error info:

```
Runtime error: The program used too much time without instructing the robot what
to do. Consider using a Wait or sync() to split long program sequences that do not
move the robot.
```

naijoug, 2017-11-29:
My issue has been solved, I must be use a wired network connection with UR3.

gavanderhoorn, 2019-06-29:
Note for potential wrid19 participants: #310 already includes related changes.

gavanderhoorn, 2019-11-23:
We'll not be improving compatibility of `ur_modern_driver` with CB3 controllers.

Please use `ur_robot_driver` instead.

closed 2019-11-23 by gavanderhoorn