ros-industrial-attic/ur_modern_driver #148: Robot begins moving erratically if left on for a long enough time period.

symbioRahul, 2017-11-13:
When running:
```bash
roslaunch ur_modern_driver ur10_ros_control.launh robot_ip:my.robot.ip
```
and leaving it on for a long enough time, the robot will start moving on its own. Is there any reason for it to be doing this? This seems like there are incorrect URScript messages being sent to the robot arm. 

This problem seems to occur in v3.3.2.xxx and an earlier v3.x version.

gavanderhoorn, 2017-11-14:
You appear to be using the `ros_control` `hardware_interface`. There are some issues with that (see the tracker) but it should not start moving on its own. Seeing as `ros_control` is involved, this could also be numerical instability or windup in the controllers.

Your post is rather vague, so we're going to need more information:

 - how long is "long enough"?
 - what is the exact configuration you are running? Please mention and / or include launch files and any config files that you have changed or tweaked.
 - what sort of motion are you seeing? Is it wildly swinging around, or is it small changes around its current pose?
 - which versions of ROS? On which OS? CPU arch?
 - does this also happen with `ur-sim`?

and if possible: run a wireshark capture to try and capture the network traffic. Be sure to trim it down to a reasonable size (ie: some time before and after the problem occurs).

miguelprada, 2017-11-14:
It may be similar to the issues we're having with one of our robots. I think it is related to the `speedj` commands _sometimes_* responding with delay (see #55 and #114). This makes the PID used by the `vel_based_pos_traj_controller` unstable and the arm will slowly start oscillating.

If that's the case, a quick workaround is to use the `pos_based_pos_traj_controller`, although its performance is a bit worse. Could you try this, @symbioRahul, and tell us if your problem goes away?

(*) We have two robots in our lab with the exact same Polyscope version and this issue is showing only in one of them. I have yet to find some time to figure out what's going on here.

gavanderhoorn, 2017-11-14:
Thanks @miguelprada: I was actually thinking of those issues as well.

We'll have to wait for @symbioRahul to provide us with more information.

symbioRahul, 2017-11-15:
Some very interesting updates/ finds:

-  *Long time* is actually whenever my wifi disconnects, the robot starts to move is "shuttering" motions. I can now repeat this bug by disconnecting the wifi connection. 

- When we EStop the robot the last commands on the pendant's log **before** the estop commands is `program jSpeed paused` instead of what I'd expect (which is `program jSpeed stopped`). When the Estop lifted and the motors turned on, the robot slams into itself. 

I'm wary of continuing to test reproduce these failures as it does damage the robot arm whenever it slams into itself. If you still want I could run Wireshark and probe the controller a bit more.

gavanderhoorn, 2017-11-15:
@symbioRahul wrote:
> -  *Long time* is actually whenever **my wifi disconnects**, the robot starts to move in "shuttering" motions collapsing onto itself. I can now repeat this bug by disconnecting the wifi connection pretty reliably. 

I'm not saying we're not going to try and do something about the problem you're reporting, but in general I would say that I would *strongly* discourage using any sort of wireless connection between you and an industrial robot.

No matter what sort of external motion control interface you use, anything that can potentially randomly drop packets and / or lose connection completely is really asking for trouble.

Again, we'll try to see whether there is something we can do about this, but if you're using a velocity interface (`speedj(..)`) and you're not supplying the robot with new setpoints 'fast enough' (or *ever*), then the changes of it doing things you didn't intend are rather high.

gavanderhoorn, 2017-11-15:
Note btw that my comments about a wireless segment in the critical path between you and the robot applies no matter whether you are using ROS / `ur_modern_driver` or not: even just sending urscript statements to `TCP#30003` manually over wireless is brittle.

symbioRahul, 2017-11-15:
@gavenderhoorn
I will agree with you there that you should not connecting via wifi in an industrial setting and that would be the best way around this issue altogether. However, would there happen to be a specific reason for why this robot would be moving in an absence of computer connection? Especially when there were no movement commands given to the robot arm after roslaunch and before disconnection?

miguelprada, 2017-11-16:
@symbioRahul If you start up the robot _driver_ with the roslaunch command above, it doesn't matter whether you send trajectory commands or not, the driver will start sending `speedj` commands to the robot right away. They will only try to keep the robot at its initial position, using a PID to achieve this.

In my experience, the smallest deviation in the robot position (probably just sensor noise) will make the PID apply tiny corrections, and these will eventually make the robot unstable due to the delays.

The behaviour when the disconnection happens is quite unpredictable, IMHO, since it depends on whichever `speedj` command was last sent.

miguelprada, 2017-11-16:
I just uploaded a small tweak to the _ros_control_ launchfiles which allows you to select which controllers to start at launchtime. Could you try to checkout [this branch](https://github.com/miguelprada/ur_modern_driver/tree/controller_launch_args) and launching as this?

`roslaunch ur_modern_driver ur10_ros_control.launch robot_ip:=my.robot.ip controllers:="joint_state_controller force_torque_sensor_controller pos_based_pos_traj_controller" stopped_controllers:="vel_based_pos_traj_controller"`

gavanderhoorn, 2017-11-16:
@symbioRahul: I think the comment by @miguelprada should clarify what is going on.

The velocity based config of the `ros_control` version of the driver makes `ros_control` close the position loop by running actual PID controllers. If the controllers send a velocity command and you then disconnect them from the actuators, the last commanded velocity will be maintained by the robot controller.

gavanderhoorn, 2018-02-02:
Closing this due to no activity.

@symbioRahul: if you feel this is something that still needs more discussion, please re-open.

closed 2018-02-02 by gavanderhoorn