ros-industrial-attic/ur_modern_driver #151: Wrong messages received by the driver

SrdjanStevanetic, 2017-11-24:
Dear Thomas and all,

I am writing this comment with regard to the closed issue #128 .
I have done some additional testing and I am still facing the problem I explained above. Let me explain in a bit more detail again what is going on.
Setup: We use 2 robots (KUKA and UR3) together with some other IoT devices like conveyor, Tablet, etc. We have a windows PC with VM-Ware server hosting some VMs (e.g. for controlling KUKA). That windows PC is connected over a gigabit router and then a switch to the real robot controllers from KUKA and UR3. Beside the windows PC I have installed the ur modern driver in a separate linux 14.04 machine and connected it to the switch using a cable connection. Since we discussed above that running the ur driver in a VM might cause a problem I installed it in a separate linux machine. In a scenario that we use, the KUKA robot transports some object to the ur3 robot which should do screwing. During the communication with the UR3, I got some messages that have strange length (like 10736546524, 0, -10736546356, etc.) and their content is not the right one, meaning that they are not valid. I am printing these messages in the robot_state_RT.cpp file in the part where the messages are unpacked. For example:

good messages (btw. we use the firmware version 3.3):
Message length: 1060.000000
Time: 3871959.552000
Actual joint positions: -3.203961,-1.570940,-1.570617,-1.570820,1.570650,0.571769,0.571769
Protocol version: 3.300000
Message length: 1060.000000
Time: 3871959.560000
Actual joint positions: -3.203949,-1.570904,-1.570652,-1.570832,1.570674,0.571769,0.571769
Protocol version: 3.300000

bad messages:
Message length: 1071499937.000000
Protocol version: 3.300000
Message length: 0.000000
Protocol version: 3.300000

Message length: 1073310141.000000
Protocol version: 3.300000

Message length: 1081057280.000000
Protocol version: 3.300000

The number of wrong messages in the test that we did is 2237 out of totally 93991 messages. In another test is was 776 out of 88732. So these bad messages are filtered out in the part that checks the message length in the robot_state_RT.cpp file. The order in which bad messages appear is random and not cyclic.

So, do you have any idea what can be the problem we are facing with? Btw, I also did tests with only the UR3 robot, without another robot and devices and the same problem happens. Could it be that some packets are lost in the communication among the real ur3 and the linux machine where the driver is running? Since I checked in the unpack(...) function in the robot_state_RT.cpp file that the messages are wrong, it means that those wrong messages are also sent from the real robot, right?
Many thanks.

Regards,
Srdjan

SrdjanStevanetic, 2017-11-27:
Hi all,

I have also tested only the UR3 robot with a direct wired connection to it and the same problem happens again, even before the robot do any movement. Just to mention that my Ubuntu 14.04 is installed on a USB 3.0.

Regards, Srdjan

mintos002, 2018-06-19:
I'm haveing the same issue, only the 5% of the packets receaved in the Real Time Communication class have the correct size, but the 95% do not! But if I capture the packets with a sniffer, all packets have the correct size.
Any updates of this issue?

Thanks!

closed 2019-11-23 by gavanderhoorn