ros-industrial-attic/ur_modern_driver #152: ur_ros_controllers lost control of the robot after sending /ur_driver/URScript command

maverick-long, 2017-12-01:
Dear all,

I want to use ur_ros_controllers to control the robot by sending trajectory message and use /ur_driver/URScript command to control the robot move forward a little bit in Cartesian space. However, when I used both of them together, after sending the /ur_driver/URScript command, the ur_ros_controller lost control of the robot. The actionlib didn't response the command. Any thought?

By the way, I used the position control in ur_ros_contorller.

Thank you very much,
Xianchao

gavanderhoorn, 2017-12-05:
the `ros_control` code path for the position interface uses a small program on the controller to receive and process `servoj(..)` setpoints.

If you publish something to `/ur_driver/URScript` that program is stopped, your URScript snippet executed and then the controller waits.

As the 'small program' doesn't get restarted, the hardware interface has no way to communicate with the controller anymore, leading to the issue you report.

gavanderhoorn, 2017-12-05:
It's debatable, but at the moment I would say this is by-design. As the readme notes: `ros_control` continuously controls the robot, and it essentially has exclusive control over it, closing the loop using the PC. As noted in the readme:

> As ros_control continuesly controls the robot, using the teach pendant while a controller is running will cause the controller on the robot to crash, as it obviously can't handle conflicting control input from two sources. 

Going behind its back (ie: sending URScript yourself) could be compared to "using the teach pendant".

The readme continues with:

> Thus be sure to stop the running controller before moving the robot via the teach pendant

It also explains how to do that using the `controller_manager` services.

maverick-long, 2017-12-06:
I tried the work flow you mentioned and it worked. Thank you!

closed 2017-12-06 by maverick-long