ros-industrial-attic/ur_modern_driver #154: How to change default controller?

herczegzoltan, 2017-12-06:
Hello,

I would like to do some measurement with position and velocity controller but I can not change it. 
Could you help me how to change or what to do for it?

Thanks.

Best Regards,
Zoltan

miguelprada, 2017-12-06:
First of all, you should be using the driver with `ros_control` support enabled.

If you are, then you can either set a new default before launching, such as [here](https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/launch/ur5_ros_control.launch#L40), or change it at runtime calling the [`controller_manager/switch_controller`](http://wiki.ros.org/controller_manager#Services) service. You can alternatively use [`rqt_controller_manager`](http://wiki.ros.org/rqt_controller_manager), which provides a GUI for the service

gavanderhoorn, 2018-03-02:
As @miguelprada notes, only the `ros_control` infrastructure supports switching controllers on-the-fly.

Then use the provided services with the appropriate arguments.

closed 2018-03-02 by gavanderhoorn