ros-industrial-attic/ur_modern_driver #156: PolyScope's log-messages

xfr-rg, 2017-12-12:
Hello all,

I'm interested in the log-messages of PolyScope as you can see them in the lower half of the "Log"-tab. Is there any possibility to access the strings of the logs and the corresponding timestamp with ur_modern_driver? I want to write a publisher.

Thanks in advance!

gavanderhoorn, 2017-12-12:
As far as I know there are no URScript statements that allow retrieving / reading anything from the log. So that approach probably wouldn't work.

The log is however a simple text file on the controller, and seeing as it is just a Linux machine, it should be possible to `tail -f` it, either locally on the controller, or over SSH. You could then parse the lines and publish them.

This is a bit out-of-scope for the driver (as it wouldn't really be a part of the driver, but a stand-alone script/node), so I'm guessing support for this would probably have to come from a contribution.

xfr-rg, 2017-12-13:
This seems like a workable solution.
What i'm trying to do is to send URscript commands to the robot using the ur_modern_driver. 
I wrote a couple of functions to wrap URscript command e.g. get_tcp_forces
As I cant get a response when sending a URScript-statement i thought i can get the response from the logs. 
Any other solutions to receive the return value of a URScript statement with the ur_modern_driver?

best,
ralf

Thompson104, 2018-01-26:
@gavanderhoorn 
I wonder if you have some suggestions regarding how to take the return value from URScript? I was trying to get the return value of "get_joint_torques()" by publishing this command to ur_driver/URScript.

Thompson104, 2018-01-26:
@xfr-rg Hi, I'm having the same problem and I wonder if you have made progress receiving the return values from URScript?

Thank you.

gavanderhoorn, 2018-01-26:
As far as I know, reading return values from script statements sent over a TCP socket is not possible.

It's been a while since I've done anything similar though, so I could be wrong.

In any case, this is a bit off-topic for this issue tracker, and would probably best be asked on a more appropriate forum (one dealing with UR support).

@xfr-rg: I feel your original question has been dealt with in a way, so I'm going to close this issue.

Please re-open if needed.

closed 2018-01-26 by gavanderhoorn