ros-industrial-attic/ur_modern_driver #158: Joint voltages are all zero?

happygaoxiao, 2018-01-12:
I use the `v_actual_` value and write a Publish topic in _ur_ros_wrapper.cpp_ as the `joint_states` topic.
`joint_v_msg.position = robot_.rt_interface_->robot_state_->getVActual();`
But it's all zero in ros_control mode with pos_based_pos_traj_controller. I want to get the joint voltages and joint currents to calculate energy consumption. The currents' values are all correct as the  PolyScope shows. Although the voltages just change in a small scale. Actually I can't see my topic at `rqt` with `ros_control` mode
In addition, the `joint_names`  are in the wrong order as "elbow_joint", "shoulder_lift_joint", "shoulder_pan_joint", "wrist_1_joint", "wrist_2_joint","wrist_3_joint".
I have also use the action interface on /follow_joint_trajectory, the voltage of first joint is -9.7V, and others are zeros. The joint_names' order is correct. And the topic I write is shown at `rqt`
 
Maybe the data of my UR version is not well-supported for the driver? Could you help me ?
@ThomasTimm @gavanderhoorn 
----
Universal Robots Software: 3.3.2.266(Sep 05 2016)
User Interface PolyScope 3.3.2.266
Robot Controller: URControl 3.3.2.131

gavanderhoorn, 2018-01-12:
Without seeing some code it's going to be hard to say why you observe what you describe.

For now: see #74 for a PR where a `WrenchStamped` publishers was added to the `ros_control` side of things. You should be able to do something equivalent.

gavanderhoorn, 2018-01-12:
> In addition, the `joint_names`  are in the wrong order as "elbow_joint", "shoulder_lift_joint", "shoulder_pan_joint", "wrist_1_joint", "wrist_2_joint","wrist_3_joint" in `ros_control` mode

#113.

happygaoxiao, 2018-01-13:
I use this variable `v_actual_` at
 https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/include/ur_modern_driver/robot_state_RT.h#L61  Add  a `Publisher` line as
 `ros::Publisher joint_v_pub = nh_.advertise<sensor_msgs::JointState>("joint_voltage", 1);`  below this row of `ur_ros_wrapper.cpp `https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_ros_wrapper.cpp#L652.  
Then I add this below the publisher of `joint_states` (https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_ros_wrapper.cpp#L677.
```
//joint voltages
 joint_v_msg.header.stamp = joint_msg.header.stamp;
joint_v_msg.effort = robot_.rt_interface_->robot_state_->getVActual();
joint_v_pub.publish(joint_v_msg);
ROS_INFO_STREAM("Vol  = "<<joint_v_msg.effort[0]<<" "<<joint_v_msg.effort[1]<<" "
        <<joint_v_msg.effort[2]<<" " <<joint_v_msg.effort[3]<<" "
        <<joint_v_msg.effort[4]<<" "<<joint_v_msg.effort[5]<<" ");
```
With the action interface on `/follow_joint_trajectory`, the voltage of first joint is -0.094V, and others are zeros.

gavanderhoorn, 2018-01-13:
> Add  a `Publisher` line as
 `ros::Publisher joint_v_pub = nh_.advertise<sensor_msgs::JointState>("joint_voltage", 1);`  below this row of `ur_ros_wrapper.cpp `https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_ros_wrapper.cpp#L652

As you can see in #74, the `publishRTMsg(..)` is not used in `ros_control` mode.

Please check #74 carefully and modify accordingly.

gavanderhoorn, 2018-01-13:
Why are you using `JointState` msgs for this?

gavanderhoorn, 2018-01-13:
> With the action interface on `/follow_joint_trajectory` (ros_control is disabled) , the voltage of first joint is -0.094V, and others are zeros.

In this mode your changes should be running.

Why the values are incorrect, I cannot say at this time.

I suggest to either debug using `gdb` to see what whether `robot_.rt_interface_->robot_state_->getVActual()` returns correct values, or add a `printf(..)` (or similar) at a suitable point.

happygaoxiao, 2018-01-13:
I have used the `ROS_INFO_STREAM()` to print the values of `robot_.rt_interface_->robot_state_->getVActual()` . With the action interface on /follow_joint_trajectory (no `ros_control`), it's the same as the voltage of first joint is -0.094V, and others are zeros.
Maybe the data stream is not sent as the order of  the `ur_modern_driver`?

Sorry about the `JointState` msgs. I just want to finish my experiment quickly.  I will choose a suitable ROS msgs and learn how to get these values when `ros_control` is active from your suggestion.

Thanks a lot for your help!

gavanderhoorn, 2018-01-13:
If you can capture some traffic with Wireshark I can take a look, but afaik deserialisation of `3.3.2` should be ok.

gavanderhoorn, 2018-03-02:
Closing due to inactivity.

If you've figured something out in the meantime, we'd still be interested to hear about it @happygaoxiao.

closed 2018-03-02 by gavanderhoorn