ros-industrial-attic/ur_modern_driver #161: Problem real robot(UR3) error C204A3 when I trying Moveit! execute.

Purdiding, 2018-02-27:
When I trying to moveit! execute with the real UR3 robot, it just works only a few seconds, and then I got a safety Message:
Protective Stop C204A3

I just follow this tutorial, in kinetic 1.12.12.

What's the problem with my robot?

gavanderhoorn, 2018-02-27:
Please provide us a bit more information:

 - ur polyscope and system software version
 - which tutorial is "this tutorial"?
 - are you using a wired or wireless network connection?
 - are you using a virtual machine to run ROS?

gavanderhoorn, 2018-02-27:
To see whether you're running into contention for resources, see if Zagitta/ur_modern_driver#9 works any better.

Purdiding, 2018-02-28:
Thank you.

The UR polyscope version is 3.5.2.

sorry about the tutorial information, when I edit the comment, the link is gone... 
Anyway, the tutorial is [https://github.com/ThomasTimm/ur_modern_driver](https://github.com/ThomasTimm/ur_modern_driver), 
```
roslaunch ur_modern_driver ur3_bringup.launch robot_ip:=xxx.xxx.xxx.xxx
roslaunch ur3_moveit_config ur3_moveit_planning_execution.launch
roslaunch ur3_moveit_config moveit_rviz.launch config:=true
```
When I connecting with UR3, used the wireless network.
The ROS PC is not a virtual machine.

Please help.
Thank you.

gavanderhoorn, 2018-02-28:
> The UR polyscope version is 3.5.2.

I'm not sure this driver supports that version properly already, so there could be issues when trying to use it.

To determine whether or not this is actually a resource issue (ie: your PC cannot sustain the high update rate used by the nodes in this repository, try and see whether Zagitta/ur_modern_driver#9 improves the situation.

I'm also not sure whether `3.5.2` has already been tested with that PR either though.

simonschmeisser, 2018-02-28:
I would recommend to use a cable instead of wireless network, especially when you try to use real time control

gavanderhoorn, 2018-02-28:
O, thanks @simonschmeisser: I'd completely missed the "used the wireless network".

@Purdiding: I would recommend to avoid using wireless networks while controlling industrial robots remotely. It can be made to work, but please only do that after you've got things to work.

My recommendation to try out Zagitta/ur_modern_driver#9 still stands though, as that should work better with a wireless segment between you and the UR control box.

Purdiding, 2018-03-02:
Thank you all~~!!!

I'll try to wired network,(but still I wanna wireless enviroment... ).

And then write a feedback~~!! 

Thank you @gavanderhoorn @simonschmeisser.

gavanderhoorn, 2018-03-02:
The driver in this repository attempts real-time control with a frequency of 500 Hz. Typical wireless networks will not be able to handle that, leading to time delays in control signals whenever a pkt is lost. After that, the controller on the ROS side tries to 'catch up' leading to the safety error you encountered.

gavanderhoorn, 2018-03-02:
I'm going to close this issue, as it's most likely a problem with using a wireless network. We strongly recommend to use a wired connection.

If you run into any more trouble with a cable, feel free to re-open.

closed 2018-03-02 by gavanderhoorn