ros-industrial-attic/ur_modern_driver #162: The URScript topic

tianxiang84, 2018-03-01:
Hi Thomas and all,

I am trying to use the URScript topic to control a Robotiq gripper, but somehow cannot make it work. If anyone has suggestion, please kindly let me know.

The following script works on the UR teach pad, when it is run, the gripper can be activated and then close as expected.

- socket_set_var(“ACT”, 0, “1”)   # Deactivate gripper 1
- sleep(0.5)     # Wait for 0.5 sec to write the registration “ACT”
- socket_set_var(“ATR”, 0, “1”)   # Set the gripper as normal
- sleep(0.5)     # Wait for 0.5 sec to write the registration “ATR”
- socket_set_var(“ACT”, 1, “1”)  # Activate the gripper
- sleep(3.0)     # Wait for 3 sec because the gripper will move in activation
- socket_set_var(“GTO”, 0, “1”)  # Stop, don’t move
- sleep(0.5)     # Wait for 0.5 sec to write the registration “GTO”
- socket_set_var(“POS”, 255, “1”) # Set the gripper position (0 is fully open, 255 is fully close)
- sleep(0.5)    # Wait for 0.5 sec to write the registration “POS”
- socket_set_var(“GTO”, 1, “1”)    # Move according to the POS setting
- sleep(5.0)    # Wait for 5 sec for the gripper to move
- socket_set_var(“GTO”, 0, “1”)   # Reset, don’t move when it is activated again next time
- sleep(0.5)   # Wait for 0.5 sec to write the registration “GTO”
- socket_set_var(“POS”, 0, “1”)   # Reset to fully open position
- sleep(0.5)   # Wait for 0.5 sec to write the registration “POS”
- socket_set_var(“ACT”, 0, “1”)   # Deactivate gripper 1
- sleep(0.5)     # Wait for 0.5 sec to write the registration “ACT”
- socket_set_var(“ATR”, 0, “1”)   # Set the gripper as normal
- sleep(0.5)     # Wait for 0.5 sec to write the registration “ATR”
- socket_set_var(“ACT”, 1, “1”)  # Activate the gripper
- sleep(3.0)     # Wait for 3 sec because the gripper will move in activation
- socket_set_var(“GTO”, 0, “1”)  # Stop, don’t move
- sleep(0.5)     # Wait for 0.5 sec to write the registration “GTO”
- socket_set_var(“POS”, 255, “1”) # Set the gripper position (0 is fully open, 255 is fully close)
- sleep(0.5)    # Wait for 0.5 sec to write the registration “POS”
- socket_set_var(“GTO”, 1, “1”)    # Move according to the POS setting
- sleep(5.0)    # Wait for 5 sec for the gripper to move
- socket_set_var(“GTO”, 0, “1”)   # Reset, don’t move when it is activated again next time
- sleep(0.5)   # Wait for 0.5 sec to write the registration “GTO”
- socket_set_var(“POS”, 0, “1”)   # Reset to fully open position
- sleep(0.5)   # Wait for 0.5 sec to write the registration “POS”_

Then I was trying to publish these line by line to the ur_driver/URScript topic. The gripper does not do anything.... I tried to publish a simple "power off" to the topic, and it works, the power is turned off, so seems to me topic is good.

I checked the log on the teach pad, whenever I send the "socket..." message, there is a "Program socket_set_var started", immediately followed by a "Program socket_set_var stopped". So I guess it is the timing issue? In order for it to work, it should not stop so quickly?

I am fairly new to ROS, so I could be doing something totally wrong, if anyone has any idea, please let me know.

Thanks.

Tianxiang

gavanderhoorn, 2018-03-02:
Sending a msg to the `urscript` topic essentially causes a new program to be started on the controller. Sending `N` individual statements, as you are doing, thus causes `N` programs to be started, one after the other.

You'll probably have to put all statements in a single program and send that in one message.

> I am fairly new to ROS, so I could be doing something totally wrong, if anyone has any idea, please let me know.

this is not really ROS-related: try sending the same script lines using `ncat` to port `30002` or `30003` manually, it will probably result in the same behaviour.

hansondon, 2018-03-02:
@gavanderhoorn I think sending the whole program at once could work here. But I have a related question that what if I do have to send the program line by line in order to be able to stop sending the script under certain conditions? I happen to be implementing an action that can do exactly this but I hope to give users the freedom to cancel the script at any moment.

But it seems to me that when using the URScript topic, you never know the when the command you send will be finished because there is no feedback. If this is a service that can return true when the script is finished, there should not be any problem sending it line by line.

carlosjoserg, 2018-03-02:
Hi @tiangxiang84, just saw your post on dof and here as well. Yes, you need to send a string containing all your script at once. If you use `std::string` you can use the `+` to add new lines and keep the code readible, and the filling the data in std msgs/String as defined in that topic.

@hansondon You can always set/unset IOs on the robot via the script you send, which are already exposed by the driver, so you can sync/communicate to the ROS side that your script did something or not

hansondon, 2018-03-02:
@carlosjoserg Ah! That's a good workaround. I'll need to set/unset IOs before and after every actual command, though. But it solves my problem. Thanks!

gavanderhoorn, 2018-03-02:
There's no way to detect whether "the script" has finished. That is a limitation of the infrastructure provided by UR, not of the driver btw.

tianxiang84, 2018-03-02:
Thanks everyone for the response. It is helpful.

So I tried to put these lines in a talker.cpp:

/////////////////////////////////////////////////
    std_msgs::String msg;
    std::string cmd_str;
    int count = 1;
    
    while(count <= 2){
      ROS_INFO("Begin to send a message");
      //cmd_str = "power on";
      cmd_str = "socket_set_var(\"ACT\", 0, \"1\")";
      
      //cmd_str += "socket_set_var(\"ATR\", 0, \"1\")\n";
      //cmd_str += "sleep(0.5)\n";
      //cmd_str += "socket_set_var(\"ACT\", 1, \"1\")\n";
      //cmd_str += "sleep(3.0)\n";
      //cmd_str += "socket_set_var(\"GTO\", 0, \"1\")\n";
      //cmd_str += "sleep(0.5)\n";
      //cmd_str += "socket_set_var(\"POS\", 255, \"1\")\n";
      //cmd_str += "sleep(0.5)\n";
      //cmd_str += "socket_set_var(\"GTO\", 1, \"1\")\n";
      //cmd_str += "sleep(5.0)\n";
      //cmd_str += "socket_set_var(\"GTO\", 0, \"1\")\n";
      //cmd_str += "sleep(0.5)\n";
      //cmd_str += "socket_set_var(\"POS\", 0, \"1\")\n";
      //cmd_str += "sleep(0.5)\n";

      msg.data = cmd_str;
      chatter_pub.publish(msg);
      ROS_INFO("Message is: %s", msg.data.c_str());
      count++;

      ROS_INFO("End of sending this message");

      ros::spinOnce();
      loop_rate.sleep();
    }
//////////////////////////////////////////////////////////////////////////////////////

On the console, I received:
[ INFO] [1519997892.221843195]: Begin to send a message
[ INFO] [1519997892.221955734]: Message is: socket_set_var("ACT", 0, "1")
[ INFO] [1519997892.221986258]: End of sending this message
[ INFO] [1519997902.221929296]: Begin to send a message
[ INFO] [1519997902.222048838]: Message is: socket_set_var("ACT", 0, "1")
[ INFO] [1519997902.222106172]: End of sending this message

On another terminal where I used rostopic echo ur_driver/URScript, I received:

sdrwcs@ia3:~$ rostopic echo ur_driver/URScript
data: "socket_set_var(\"ACT\", 0, \"1\")"
---

So the problems are:
1. I tried running "socket_set_var("ACT", 0, "1")" alone on the UR teach pad (just this single line in a brand new program file), this successfully deactivate my gripper. Then I activated my gripper again from the teach pad, and sent the message to the ur_drive/URScript using the above c++ script, the gripper remains activated. If even this single command does not work, I feel sending the entire script will not work either?

2. In the C++ script, I looped twice, so I saw in the console there are 2 [begin messages]...[end messages], which makes sense. But on the rostopic echo terminal, I only got 1... and this is from the 2nd one. What did I do wrong that I don't get the first published string?

3. What I want to send is "socket_set_var("ACT", 0, "1")", and on the rostopic echo terminal, I got "socket_set_var(\"ACT\", 0, \"1\")". So the double quote become \", will this be a problem?

I tried the same script to publish the message "power off". If I loop twice, the power was successfully turned off the 2nd time. So I think the script can successfully publish messages to the URScript topic. But still here the problem is if I loop only once, there is no power off response.

I feel I may be making some simple mistake that I cannot see. Please let me know where I did wrong.

Thanks.

Regards,
Tianxiang

felixvd, 2018-08-09:
Not a very qualified comment because I haven't used URScript a lot, but I know that I sent this example program to the /robot_name/URScript topic successfully. Note the wrapping function definition, which might be required:

```
def move_forward_10cm():
    start_pos=get_forward_kin()
    target_pos=pose_trans(start_pos,p[0.0, 0.0, 100.0, 0.0, 0.0, 0.0])
    movel(pose_trans(p[0.0,0.0,0.0,0.0,0.0,0.0], target_pos), a=1.2, v=0.25)
end

```

ctaipuj, 2018-09-11:
Hi @tianxiang84 I saw your post few days ago and based on your comments and @carlosjoserg 's suggestions, I developed this C++ class to control the Robotiq's 85 Gripper through  UR3 controller using URScript ROS topic. I 've tested it in a sample node and the gripper moves and works fine... You can find the source code/package [here](https://github.com/ctaipuj/ur3_gripper_control). Notice that the robotiq URcap must be installed in the controller and that @felixvd comment is valid, it is required to use the def/end statement (Almost every time). I'm not expert neither in ROS nor C++ so if anyone has any suggestion/recommendation please let me know.
Nico.

felixvd, 2019-04-04:
This is a popular google result, so in case someone can use it, I'll just link [this header file](https://gist.github.com/felixvd/4d5db1c297a3435cabb40c857eb6ead5) that waits for the UR controller to complete a program. It will work as soon as #247 is merged.

As it says, the ur_modern_driver script does not stop, so can only check if your own URScripts that you sent to the urscript topic have completed, but it might be useful for someone.

Giessen, 2019-04-15:
@ctaipuj could you give a valid link please~~~ the old is dead. Thanks (:

ctaipuj, 2019-05-23:
Hello @Giessen @DavidYaonanZhu, sorry for the delayed response. I just fixed [the link](https://github.com/ctaipuj/luc_control/tree/master/robotiq_85_control) to the repo. Sorry but currently this package doesn't have a readme file. Please check the doc directory where you can find the class documentation.

gavanderhoorn, 2019-06-29:
Seeing as this issue has gone somewhat off-topic (from "how to use urscript topic" to "how to control robotiq grippers") I'm closing this.

closed 2019-06-29 by gavanderhoorn