ros-industrial-attic/ur_modern_driver #163: Ros controller for online inverse kinematic

paolovigkiller, 2018-03-02:
Hi Thomas and all,
i want use ros control to send a single point of my trajectory because i have to implement online inverse kinematic. Can i do that or i have to send overall trajectory? If I can, i have to publish on which topic?
Thank you, i hope i was clear.

gavanderhoorn, 2018-05-08:
I'm slightly unsure as to what you're asking exactly.

If you're trying to do on-line trajectory generation and want to essentially *stream* trajectories to the driver, one point at a time, I'm afraid that is currently not supported.

I see two ways of implementing this:

 - use the `/joint_speed` topic: this takes joint velocities so that may require you to perform your own IK and probably requires some work with Jacobians.
 - use the `ros_control` side of the driver and exploit the trajectory replacement (or: blending) functionality of the `joint_trajectory_controller` (ie: send longer trajectories than a single point, but replace them at a sufficient rate)

gavanderhoorn, 2018-05-08:
I'm going to close this as it's not an actual issue with the package in this repository.

Could I please ask you to ask these sort of questions on ROS Answers in the future? We'd like to reserve the issue tracker for reports of bugs and enhancement requests.

Do please feel free to keep commenting.

closed 2018-05-08 by gavanderhoorn