ros-industrial-attic/ur_modern_driver #164: UR3 position and velocity control with ros_controllers

littlejohnyang, 2018-03-14:
I am using UR3 arm for my project. I have implemented my own planner and want the arm to follow my computed trajectory with position and velocity value controller by the ros_controller. I notice the best bet is to use the PosVelAccJointInterface from ros_controller. However, I can't find a way to set the joint speed at each joint. It looks like the servoj is able to set the target position. I'm curious what is the way to set joint speed?

littlejohnyang, 2018-03-14:
I noticed there is a "servoj" URScript that we can call. However, it's like setting pos and time. If I understand correctly, the ros_controllers gives pos and velocity and managed to time like your doTraj() function. My experience with other arms is that I can just set pos and vel for each joint, and it should work. However, I don't know how to set the velocity with UR3 at the same for position movement.

Gritzou, 2018-05-07:
" servoj or speedj is only option" means ur3 cannot follow trajectroy produced by your own planner?
or it can  only execute trajectory without velocity control?
besides,what about ur5，i'm considering to buy  a robot for experiment.(ur3/ur5)
i want to use moveit  drive the robot both with position ,velocity and acceleration
my tutor prefer UR3,but i wonder whether UR3 fit my demand
what's your advice?
thanks for reading/