ros-industrial-attic/ur_modern_driver #165: Interface with UR Dashboard Server

vickailiu, 2018-03-16:
as described here: https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/dashboard-server-port-29999-15690/

The UR Dashboard Server interface could provide more informations such as robot state and control such as showing/dismiss the popup on dashboard.

Will the new driver integrate this part in, or it deviates the motivation of this driver?

vickailiu, 2018-03-16:
I see some work done like #99 .

gavanderhoorn, 2018-03-18:
What sort of functionality are you looking for that the dashboard could provide?

> The UR Dashboard Server interface could provide more informations such as robot state

I believe the dashboard server makes use of the same datastreams as are used by the driver to determine robot state (ie: booting, calibrating, etc). Could you be a bit more specific?

> and control such as showing/dismiss the popup on dashboard.

Showing popups may be interesting. See #5.

vickailiu, 2018-03-19:
@gavanderhoorn I must admit that I didn't do my homework -- reading through issue lists well. For robot state, I mean "robotmode", "programState" and "safetymode". They provide valuable informations for controlling the robot via programs.

gavanderhoorn, 2018-03-19:
All of those are part of the datastreams received by the driver, it may just be lacking publishers for them.

#143 may already cover some of them with the `RobotStatus` message.

gavanderhoorn, 2018-05-08:
@vickailiu: I'm going to close this. Not because I don't believe there is any value in your question, but because this is not an actionable issue at the moment.

As I wrote in my earlier comment(s): the specific pieces of info you mention should already be part of the datastream(s) that the driver receives from the controller. If they are not already published, then adding a publisher for them (or some other way in which they could be retrieved) would seem to be the best course of action.

Afaict, the dashboard server seems to be another way of retrieving the same data, but adding support for that would require much more extensive changes than reusing the existing infrastructure.

gavanderhoorn, 2018-05-08:
Do please feel free to keep commenting on this issue btw.

closed 2018-05-08 by gavanderhoorn

vickailiu, 2018-05-10:
@gavanderhoorn agree.