ros-industrial-attic/ur_modern_driver #166: CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):

nazir-ust, 2018-03-18:
Each time I build the `ur_modern_driver` package I get this warning

`catkin_package() DEPENDS on 'ur_hardware_interface' but neither
  'ur_hardware_interface_INCLUDE_DIRS' nor 'ur_hardware_interface_LIBRARIES'
  is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:110 (catkin_package)`

When I try to move my dual arm UR10 with ros_control and moveit, the planner succeeds, but the robot does not move. Do you think this warning can in anyway be related to this issue of mine?

gavanderhoorn, 2018-03-18:
> Do you think this warning can in anyway be related to this issue of mine?

No. This is a CMake configuration phase warning that has no influence on runtime behaviour of the driver.

gavanderhoorn, 2018-03-18:
It is a problem with the build script though (it should not `DEPEND` on a library that it builds itself), so keeping this open to track it.