ros-industrial-attic/ur_modern_driver #173: Why UR stops?

inksci, 2018-05-11:
When I use:
```
speedj([0, 0, 0, 0, 0.5, 0], 2)
```
Actually, I hope that wrist_2_joint of the robot will keep running at a velocity of 0.5 (until it reaches its limitation).
However, the truth is that this joint will stop after a short time.
If I use this one:
```
speedj([0, 0, 0, 0, 0.5, 0], 20)
```
The joint will be running more slowly and also stop after a short time.

What should I do to let the joint keep running at a certain velocity rather than stop time by time?

Thanks a lot!

gavanderhoorn, 2018-05-11:
I'm a little bit confused about your issue: can you clarify how this is related to the package in this repository?

This tracker is not a discussion forum for URScript related topics, but reserved for issues with the `ur_modern_driver` package.

inksci, 2018-05-13:
As the use of the topic: '/joint_speed', the information in README mentions that it is suitable for visual servoing.

We publish the topic '/joint_speed' by setting the joint speeds and accelerations, as well as the time t. As a result, the robot will run at such speeds and accelerations, but it will stop (or block) after the time t.

In my opinion, for the application of visual servoing, I think that the robot should keep running at such velocity and acceleration until the velocity and acceleration are changed next time according to the real-time feedback.

In one word, I need the robot keep running at the velocity and acceleration set by me rather than stop after a given time. How can I achieve that with the help of this ur_modern_driver package, and thanks a lot!

ThomasTimm, 2018-05-13:
There is a safety implementation, in that if no data is published to the driver for a certain period of time (300 * 8ms afair), the driver stops the robot.
This way, your robot doesn't reach its hard limit if the commanding ROS node crashes.
Just send your command continuously (as you'd probably do when doing visual servoing), and it should work as you'd expect.
Please re-open the issue if this doesn't solve your problem.

closed 2018-05-13 by ThomasTimm

inksci, 2018-05-14:
Thanks a lot!