ros-industrial-attic/ur_modern_driver #174: Example code for publishing urscript command to the robot

lakshmip001, 2018-05-22:
I need to sent a movep command using /urscript topic. I am new to using this ur_modern_driver .please provide an example code for publishing movep command on the topic.

gonzalocasas, 2018-05-22:
It's rather straightforward:

1. Start advertising on `/ur_driver/URScript` (using `std_msgs/String` message type)
2. Publish messages to it where the data payload is your URScript string, i.e. `movep(...)`
3. Party 🎆

lakshmip001, 2018-05-23:
Thanks . I will do.

lakshmip001, 2018-05-23:
Does it work only with real robot or can I work with the UR simulator?

lakshmip001, 2018-05-23:
Thanks gonzalocasas. I did write cpp code to publish movep() command on URScript topic . i can see publisher publishing data, when I do rostopic info. It shows a publisher: talker and subscriber: ur_driver. But I dont see robot movement. i added <exec_depend>ur_modern_driver<exec_depend>. suggest how I can move robot?

gonzalocasas, 2018-05-25:
@lakshmip001 please don't post duplicated issues. This is a duplicate of #175.

gavanderhoorn, 2018-11-01:
As this is a duplicate of #175 (and the original question seems to have been resolved), I'm closing this.

closed 2018-11-01 by gavanderhoorn