ros-industrial-attic/ur_modern_driver #175: problem with movement of ur robot with urscript

lakshmip001, 2018-05-25:
Hi, I published URScript command movep[(q,)] using /ur_driver/URScript topic by senting a string message. I can see the publisher publising the msg and urscript subscribed to it. But I am unable to move the robot . Please suggest me what would be the problem.

SteveGrehl, 2018-05-25:
Since this may is related to the communication between ROS and UR, can you post information about your versions?
I run into a similar problem with `kinetic` and `UR 3.5.2` on my UR5 (see: https://github.com/Zagitta/ur_modern_driver/issues/18). It was fixed by adding a terminal symbol to the string. I suggest you try the fork of @Zagitta found here:  https://github.com/Zagitta/ur_modern_driver

For debugging purposes you can use a script to test, if the errors is in the ROS-interface. 
[bypass_ur_script.txt](https://github.com/ThomasTimm/ur_modern_driver/files/2038318/bypass_ur_script.txt)
I also recommend to first test with a Virtual Machine from UR found here: https://www.universal-robots.com/download/?option=37648#section16597

gonzalocasas, 2018-05-25:
I have only tested this with @Zagitta's fork anyway, so I don't know if there's an issue in this repository that prevents this from working. And +1 to @SteveGrehl 's suggestions of adding extra info/context.

lakshmip001, 2018-05-28:
I am using ROS Kinetic with UR Controller version 3.5.1. I used the code talker directly to publish command on /urscript topic without using socket communication. I am able to move robot arm but same as expected and there is delay in the program.I will try your method using socket programming.

gonzalocasas, 2018-05-28:
> I am able to move robot arm
well, that's good 👍 

> [..] but same as expected 
I don't understand what you mean.

lakshmip001, 2018-05-28:
I mean I am directly publishing movep() command on /URScript topic and giving displacement along y-direction. I checked the motion before sending the command the robot y displacement towards front direction but robot displaces downwards. And when run the command rosrun the robot doesnt move . Robot stands still and when I click on keyboard interrupt ctrl+c it moves. And for code the attached by @SteveGrehl uses a subscriber while there exists two subscribers and no publisher. I tried defining publisher but still I am unable move the robot arm according to @SteveGrehl code.

lakshmip001, 2018-05-28:
Kindly please find the attached text code.
[talker.txt](https://github.com/ThomasTimm/ur_modern_driver/files/2044638/talker.txt)
nvidia@tegra-ubuntu:~/catkin_ws$ rostopic list
/attached_collision_object
/clicked_point
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/follow_joint_trajectory/cancel
/follow_joint_trajectory/feedback
/follow_joint_trajectory/goal
/follow_joint_trajectory/result
/follow_joint_trajectory/status
/initialpose
/joint_states
/joint_temperature
/move_base_simple/goal
/move_group/cancel
/move_group/display_contacts
/move_group/display_planned_path
/move_group/feedback
/move_group/goal
/move_group/monitored_planning_scene
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/recognized_object_array
/rosout
/rosout_agg
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/rviz_tegra_ubuntu_25076_3945433862842214962/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz_tegra_ubuntu_25076_3945433862842214962/motionplanning_planning_scene_monitor/parameter_updates
/tf
/tf_static
/tool_velocity
/trajectory_execution_event
/ur_driver/URScript
/ur_driver/io_states
/ur_driver/robot_status
/wrench
I meant the robot should be in motion immediately I publish commands and stop when I press keyboard interrupt. But the robot executes after some time after rosrun is executed and contiunes even when I press keyboard interrupt ctrl+C. Sometimes code start executing  even after I press ctrl+C.  As seen in the code I am displacing arm only y-direction with respect to the first point but the robot moves in other direction

lakshmip001, 2018-06-06:
I executed a program using movep() on teach pendent I dont get error and it executes properly while program the same code using /URScript by publishing commands it keeping showing circle radius is too small or invalid joints . Can I know what is the problem and How can I proceed further?

gavanderhoorn, 2018-11-01:
Is this still an issue for your @lakshmip001 ?

The `URScript` topic can be a bit difficult, and I'm not sure using it for motions is the best thing to do.

One thing I can suggest is to wrap all statements in a program (so `def myProg: ... end`) instead of sending single lines. That tends to work better.

gavanderhoorn, 2018-11-01:
I'm going to close this issue, but if it's still a problem then please feel free to re-open @lakshmip001.

closed 2018-11-01 by gavanderhoorn