ros-industrial-attic/ur_modern_driver #176: How to use ur_modern_driver in gazebo

geweihgg, 2018-05-31:
Hi,
I want to use ur_modern_driver to publish joint speed commands to control the ur5 arm in gazebo, so I clone this repository and catkin_make it. Then I rewrite the ur5.launch file to load ur_modern_driver/config/ur5_controllers.yaml, and spawn it. But some errors occur like this:
[image]
Can you help me?
Um... Maybe a clear tutorial on how to use ur_modern_driver with ur5 in gazebo is perfect.
Thank you very much.
GeWei

gavanderhoorn, 2018-05-31:
The driver cannot be used with Gazebo, as it doesn't simulate a UR controller, but only a robot mechanism.

> Um... Maybe a clear tutorial on how to use ur_modern_driver with ur5 in gazebo is perfect.

It would just state that it will not work, so I doubt that would be very useful.

gavanderhoorn, 2018-05-31:
You might have more luck with `ursim`, a 'simulation' provided by Universal Robots themselves.

That can be used with `ur_modern_driver`. Just connect to `127.0.0.1` after you've started `ursim`.

closed 2018-05-31 by gavanderhoorn

geweihgg, 2018-05-31:
Hi, @gavanderhoorn ,
Thank you very much, I'll try it~
GeWei

geweihgg, 2018-06-01:
Hi, @gavanderhoorn ,
I have one more question, can I use velocity control command to control UR in gazebo? Or is it simple to do this? All I want is to control ur5 in gazebo with velocity control command. But I don't know whether it's hard or not.
Can you give me some advice?
GeWei

gavanderhoorn, 2018-06-01:
Using `gazebo_ros_control` you could probably configure a `velocity_controllers/JointGroupVelocityController` (for all joints) or a `velocity_controllers/JointVelocityController` (for individual joints).

You should then be able to provide that with appropriate input and control it.

gavanderhoorn, 2018-06-01:
Btw:

> All I want is to control ur5 in gazebo with velocity control command.

be careful with sentences that start with `All` or contain the word `just`: the things that follow or accompany those words are often not as simple, easy or straightforward to do as is implied by using those words.

geweihgg, 2018-06-01:
Hi, @gavanderhoorn ,
Thank you for your advice~
^-^
GeWei

geweihgg, 2018-07-05:
Hi, @gavanderhoorn ,
I want to implement a velocity interface based on position controller in gazebo, but I don't know how to implement it. So I wonder how the ur_modern_driver solve this, can you tell me how the ur_modern _driver works? By integrating the velocity to position or some other methods?
Thank you very very much~
GeWei

gavanderhoorn, 2018-07-05:
> So I wonder how the ur_modern_driver solve this, can you tell me how the ur_modern _driver works? By integrating the velocity to position or some other methods?

The UR controller exposes a velocity control interface (ie: you can directly control joint velocity). That is what the driver uses.

geweihgg, 2018-07-05:
Hi, @gavanderhoorn ,
Yes, but the velocity control interface is only for the hardware, and I want to control UR in gazebo with velocity mode. Maybe integrating the velocity command and then publish it to the position interface is a feasible way?
GeWei

gavanderhoorn, 2018-07-05:
For gazebo you'll want to make sure you load the correct controller from `ros_control`. The files in `ur_gazebo` don't use anything from `ur_modern_driver`, but load `gazebo_ros_control`.

geweihgg, 2018-07-05:
I tried to load the velocity controller for UR in gazebo, and then I found that I should adjust the PID parameters for the controller, but I didn't know how to got the right parameters. Do you have any suggestions?
Thank you~

gavanderhoorn, 2018-07-05:
No, unfortunately not, other than try to use the regular way of tuning parameters of a PID. I haven't used it myself so I cannot guide you.

But a note: as soon as you start using actual controllers (so not just transparent pass-through), tuning parameters is going to be needed there is no way around it.

geweihgg, 2018-07-05:
Oh, I see. Thank you very much.
By the way, Isn't the UR velocity control interface based on the position controller?

gavanderhoorn, 2018-07-05:
> Isn't the UR velocity control interface based on the position controller?

which "velocity control interface" are you now referring to here?

The real one robot+controller, or the simulated one?

geweihgg, 2018-07-05:
The real one robot + controller, i.e. the velocity control mode in ur_modern_driver.

gavanderhoorn, 2018-07-05:
That would be something only UR can authoritatively answer, but I would expect the answer to be "no". Why would it have to be position based? The internal controller can just look at dx/dt and increase/descrease motor current to achieve a desired rotational velocity.

geweihgg, 2018-07-05:
Ok, I think the following words confuses me (It's copied from the ur_modern_driver/README.md):
[image]
It says that the velocity based controller controls the joint position too. I'm confused about this.

gavanderhoorn, 2018-07-05:
That is a position controller implemented using velocity control.

Input: position setpoint
Output: velocity command to actuator

geweihgg, 2018-07-05:
But in the ur_modern_driver/README.md, the usage is:
[image]
So if I want to use this interface, I need to specify the velocity and acceleration in the first JointTrajectoryPoint. How about the position term? Set the position to zero? I'm not sure whether I understand it correctly.

gavanderhoorn, 2018-07-05:
You need to pass a `JointTrajectory` containing a single `JointTrajectoryPoint` with the desired velocities and accelerations for that specific control iteration. Position is ignored.

geweihgg, 2018-07-05:
Oh, I see. I think I've figured out the problem. 
So if I want to control the UR with velocity mode in gazebo, I should load the velocity controller, and tune the PID parameters. I'll try it.
Thank you for your patience~     ^-^
GeWei

gavanderhoorn, 2018-07-05:
Yes (and make sure to use the appropriate hardware interface in the xacro), but please understand that all of that has nothing to do with `ur_modern_driver`.

geweihgg, 2018-07-05:
Ok, Thank you~

nag92, 2018-08-02:
@geweihgg have you had any luck?

geweihgg, 2018-08-02:
@nag92 , yes, change the controllers from position_controllers/JointTrajectoryController to velocity_controllers/JointVelocityController is ok, and we can tune the PID parameters with rqt_gui.

nag92, 2018-08-02:
Would I be able to take a look at your code?

geweihgg, 2018-08-02:
Yes, I'll show you the code tomorrow~

Abbyls, 2019-10-25:
@geweihgg 
Hi, I am also trying to use velocity control for the ur in the gazebo. However, the velocity control cannot work well, there is always gravity pull the links falling down. And in the beginning of loading the gazebo, the robot arm can not stand up at a target pose. 
Have you encountered this problem? Could you give me some suggestions?

Thanks in advance!

geweihgg, 2019-10-25:
Hi,
    I think you can try to tune the PID params for velocity controller.
GeWei

发自我的 iPad

Abbyls, 2019-10-28:
Hi GeWei,

Could you give me some guidance for tuning the PID for the velocity controller? 
Is the velocity controller follow the gazebo_ros_controller's PID parameters?  
I really have problems tuning the velocity parameters. I found I need to change the PID parameters in the gazebo_ros_controller to make it affect the velocity controllers. And at the same time, I also need to compensate for the gravity that pulling the links down to the ground.....

ipa-nhg, 2019-10-29:
I think this issue is not related with `ur_modern_driver` but with `universal_robot` repository.

Please see the following open issues:
https://github.com/ros-industrial/universal_robot/issues/462
https://github.com/ros-industrial/universal_robot/issues/386
And the PR: https://github.com/ros-industrial/universal_robot/pull/443

So far, unfortunately, we couldn't find a proper solution for this problem, please follow and comment the previously mentioned discussions.