ros-industrial-attic/ur_modern_driver #177: Only the hz=125 is suitable for the /joint_speed

inksci, 2018-06-03:
when I use the /joint_speed to control the robot, if I set the hz=125, it works well.

However, hz=125 is a such high frequency, and it is easy to lead to unstable if my program need do many works with many threads.

I think hz=20 is enough, but how can I adjust the frequency with the ur_modern_driver?

I have tried to replace the arguments in the files (replace 125 with 20, and replace 0.008 with 0.05), but it seems not work.
```

#!/usr/bin/env python

import rospy
from std_msgs.msg import String

from trajectory_msgs.msg import JointTrajectory
from trajectory_msgs.msg import JointTrajectoryPoint

import numpy as np

def talker():
    pub = rospy.Publisher('/ur_driver/joint_speed', JointTrajectory, queue_size=10)

    rospy.init_node('talker_hz125', anonymous=True)
    rate = rospy.Rate(125)

    qvel = [0, 0, 0, 0, 0, 0]

    i_seq = 0

    while not rospy.is_shutdown():

        for i in range(6):
            qvel[i] += (np.random.rand()-0.5)*2* 0.01

        trajectory = JointTrajectory()
        trajectory.joint_names = ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']

        current_time = rospy.Time.now()
        trajectory.header.seq = i_seq
        i_seq += 1

        trajectory.header.stamp = current_time
        trajectory.points = [JointTrajectoryPoint(velocities=qvel)]
        pub.publish(trajectory)

        rate.sleep()

if __name__ == '__main__':
    try:
        talker()
    except rospy.ROSInterruptException:
        pass

```

closed 2018-06-07 by inksci

gavanderhoorn, 2018-06-07:
Can you say something about how you resolved this?

inksci, 2018-06-07:
I have solved this problem with the help of multi process.

Here are my codes:

```
#!/usr/bin/env python

from multiprocessing import Process, Queue
import os, time, random

import rospy
from std_msgs.msg import String

import numpy as np

from trajectory_msgs.msg import JointTrajectory
from trajectory_msgs.msg import JointTrajectoryPoint

import threading
import time

import numpy as np

qvel_global = [0, 0, 0, 0, 0, 0]
mutex = threading.Lock()

def talker_4():

    try:

        pub = rospy.Publisher('/ur_driver/joint_speed', JointTrajectory, queue_size=10)

        rate = rospy.Rate(125)

        qvel = [0, 0, 0, 0, 0, 0]

        i_seq = 0

        global qvel_global, mutex

        trajectory = JointTrajectory()

        trajectory.joint_names = ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']

        while not rospy.is_shutdown():

            if mutex.acquire():
                qvel = qvel_global
                mutex.release()

            trajectory.header.seq = i_seq
            i_seq += 1

            current_time = rospy.Time.now()
            trajectory.header.stamp = current_time

            trajectory.points = [JointTrajectoryPoint(velocities=qvel)]
            pub.publish(trajectory)

            rate.sleep()

    except rospy.ROSInterruptException:
        pass

def talker_4_process(q):

    global qvel_global, mutex

    rospy.init_node('talker_4_process', anonymous=True)
    threading.Thread( target = talker_4 ).start()

    try:

        
        rate = rospy.Rate(20)

        while not rospy.is_shutdown():

            qvel_data = q.get(True)

            if mutex.acquire():
                qvel_global = qvel_data
                mutex.release()
           

            rate.sleep()

    except rospy.ROSInterruptException:
        pass

def ink_control(q):

    rospy.init_node('ink_control', anonymous=True)

    rate = rospy.Rate(10)

    qvel_data = [0]*6

    while 1:

        ink = 0
        for i in range(1000):
            ink += 1

        for i in range(6):
            qvel_data[i] += (np.random.rand()-0.5)*2* 0.01

        q.put( qvel_data )

        rate.sleep()

if __name__=='__main__':
    q = Queue()
    p_talker_4 = Process(target=talker_4_process, args=(q,))
    p_ink_control = Process(target=ink_control, args=(q,))
    p_talker_4.start()
    p_ink_control.start()

```