ros-industrial-attic/ur_modern_driver #178: UR10 Dramatically Overshooting with Moveit

atyshka, 2018-06-05:
I'm working with this driver and the UR10 Moveit config and I'm running into issues where the robot dramatically overshoots its intended trajectory. Moveit makes the correct plans, but when those plans are executed about 20% of the time the robot will overshoot or miss the target pose. In most cases it keeps moving until it hits a joint boundary. I'm using the joint limited launch option, which normally constrains it from -pi to pi, but when this driver error occurs joints go past those limits to 2pi. I've tested this with a simulated robot without any issues, so I'm fairly sure the issue is with the driver and not Moveit. No errors, all the console output indicates that it has successfully reached the goal:

[ INFO] [1528232158.412900600]: Received new goal
[ INFO] [1528232158.413009902]: Trajectory received and accepted
[ INFO] [1528232158.413114134]: Translating trajectory
[ INFO] [1528232158.413197991]: Executing trajectory with 19 points and duration of 2.062s
[ INFO] [1528232158.413237870]: Uploading trajectory program to robot
[ INFO] [1528232160.515501296]: Trajectory executed successfully

I've gotten big errors, with the base joint sometimes swinging a full 180 degrees past its intended position. I'm using a wired connection, ROS Kinetic, and Polyscope 3.4

enrico-g, 2018-06-20:
Same issue here with the UR5. Have you solved it?

closed 2018-08-06 by atyshka

Anastasiamor100, 2019-02-18:
I am having the same issue, how was this solved?

enrico-g, 2019-02-18:
If I remeber correctly, I solved this by adding to the "ur_hardware_interface" node the commands:

```
<param name =" servoj_gain " type =" double " value =" 800 " />
<param name =" servoj_lookahead_time " type =" double " value =" 0.07 " />
```

The node is launched by "ur5_ros_control.launch" in my case (UR5), so you can modify that one.

A long time has passed since my last time on ROS, so take into account that this may not be the solution to your problem. I hope I can help more if needed.

ashah03, 2019-03-29:
This didn't work for me (on the UR5). Before I added this, it would eventually converge to the right location after overshooting if I ran it enough (10-20) times, but after adding it it doesn't stop until it hits joint limits. Could you share the full contents of your ur5_ros_control.launch file?

Anastasiamor100, 2019-03-29:
I noticed that it only works correctly on 100% speed

enrico-g, 2019-03-29:
Yes another thing is for sure that on your MoveIt! interface you must set a lower or equal velocity scaling value with respect to what you have on your Teachpendant. 
For example, having velocity scaling on MoveIt = 10% and 100% on teachpendant should work fine. The vice-versa will not work.

enrico-g, 2019-03-29:
Unfortunately I cannot share the full content of my ur5_ros_control.launch file because I am not working anymore on ROS and I don't have access to the material anymore. But I can ask one of my old colleagues and in case I get the file I will share it here.

ashah03, 2019-03-30:
I was running it at 10%, I'll try it out and let you know if that solves it.

ashah03, 2019-04-01:
Do you know how to change the speed on Moveit?

ashah03, 2019-04-01:
Also, that seemed to solve the issue for overshooting, but with pose goal it will occasionally start moving in random directions...any ideas?

enrico-g, 2019-04-04:
To change the Velocity scaling value on MoveIt you can do It either by code or directly on the interface. In MotionPlanning - > Planning tab - > Velocity scaling (default is 1.00, namely 100%)

enrico-g, 2019-04-04:
I don't understand. You mean that when you ask the robot to reach the target it sometimes moves in random directions?