ros-industrial-attic/ur_modern_driver #184: Support for Software 5.0 (E series)

atyshka, 2018-07-12:
Has anyone experimented with using this driver with the new e-series of universal robots? There are a couple new features such as built in RS485 and force sensor that would be really useful for our purposes, but I wanted to see if there was any support planned for this new product line. I checked the new URScript reference and I didn't see any glaring changes, so it might require little to no modification. However, I have no hardware to test on.

gavanderhoorn, 2018-09-03:
This is being worked on.

In the meantime I'd be interested to hear your experiences if you've tried it already. More input is always good.

AustinDeric, 2018-10-11:
related? https://github.com/ros-industrial/ur_modern_driver/issues/136

gavanderhoorn, 2018-10-16:
@AustinDeric: I don't get notification emails for edits, so I hadn't seen yours.

No I can't share what I have at the moment.

The parser notices that there are still bytes left after deserialising all fields it knows about. At that point it emits the error.

Going through the binary protocol in [this document](https://s3-eu-west-1.amazonaws.com/ur-support-site/16496/Client_Interface_V3.7andV5.1.xlsx) and comparing pre 5.x with 5.x should reveal the field(s) added.

dniewinski, 2018-10-17:
Currently working on this on my fork.  Seems to be working, but gets hung up on uploading the trajectory to robot.  Any suggestions?  Hoping to get time to work through this today

happygaoxiao, 2018-10-19:
I have downloaded the data structure in [Client_Interface_V3.7andV5.1.xlsx](http://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/). For UR e-series, it seems to very similar to 3.2-3.4 version. I am trying to modify this driver and test in offline-simulator and real robot

gavanderhoorn, 2018-10-19:
@happygaoxiao: @dniewinski has already done that. His branch should already work with v5.x robots and controllers.

gavanderhoorn, 2018-10-19:
See #216.

lihx108, 2018-11-02:
I solve the problem #212. ur_modern_driver can still work. if I change to remote control.
 1. In the Header, press the Hamburger menu and select Settings.
2. Under System, select Remote Control.
3. Press Enable to make the Remote Control feature available. PolyScope remains active.
Note: Enabling Remote Control does not immediately start the feature. It allows you to
switch from Local Control to Remote Control.
4. In the profile menu, select Remote Control to alter PolyScope. Note: You can return to
Local Control by switching back in the profile menu, or selecting Operator or Programmer
if a password is specified.

gavanderhoorn, 2019-11-23:
Please use `ur_robot_driver` for e-series and CB3 controllers.

Compatibility of `ur_modern_driver` with these controllers and Polyscope versions will not be worked on any longer.

closed 2019-11-23 by gavanderhoorn