ros-industrial-attic/ur_modern_driver #185: Could you give me your TCP protocol

HelloDusty, 2018-07-16:
Hi
I am trying to connect UR to vrep. Thus, I am going to connect to vrep to ur driver by tcp, and then, connect vrep to ROS environment. But, it's hard to read all the code of ur_modern_driver. Could you give me your TCP protocol from ur_modern_driver to UR.
Thanks

gavanderhoorn, 2018-07-16:
The only protocol we're using is the one created by Universal Robots themselves: see [Remote Control Via TCP/IP - 16496](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/) for the documentation.

But tbh, I don't think what you're trying to do is a good approach. Now what I think doesn't really matter, but I would not try to implement the entire UR TCP protocol just to be able to interface with VREP.

gavanderhoorn, 2018-07-16:
As there isn't anything really actionable in this ticket (or at least not for the package in this repository), I'm going to close it. Do feel free to keep commenting though.

closed 2018-07-16 by gavanderhoorn