ros-industrial-attic/ur_modern_driver #186: Controlling multiple (two) robots

raequin, 2018-08-01:
To use MoveIt! with two URs in Gazebo I created a single URDF with both arms.  In MoveIt! there are two corresponding planning groups.  This works fine in simulation.  Is it possible to control two URs from a single computer with `ur_modern_driver`?  The driver works well with MoveIt! on each of the arms when I run `roslaunch ur_modern_driver urXX_bringup.launch robot_ip:=ROBOT_IP_ADDRESS` with either of the two IP addresses.  The modified launch file pasted below yields this error, though:

`[FATAL] [1533133057.969680600]: ERROR on binding socket for reverse communication
[ur_driver_placer-4] process has died [pid 20617, exit code -5`

Here is the launch file I ran:

`<launch>
  <arg name="placer_robot_ip"/>
  <arg name="attacher_robot_ip"/>
  
  <arg name="min_payload"  default="0.0"/>
  <arg name="max_payload"  default="10.0"/>
  <arg name="servoj_time" default="0.008" />
  
  
  <arg name="max_velocity" default="10.0"/> 
  
  
    
  
  <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find hydra)/models/hydra/hydra.urdf.xacro'" />  
    

  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
  

  
  <node name="ur_driver_placer" pkg="ur_modern_driver" type="ur_driver" output="screen">
  
    <param name="prefix" type="str" value="placer_" />
    <param name="robot_ip_address" type="str" value="$(arg placer_robot_ip)" />
    <param name="min_payload" type="double" value="$(arg min_payload)" />
    <param name="max_payload" type="double" value="$(arg max_payload)" />
    <param name="max_velocity" type="double" value="$(arg max_velocity)" />
    <param name="servoj_time" type="double" value="$(arg servoj_time)" />
    <param name="base_frame" type="str" value="placer_base"/>
    <param name="tool_frame" type="str" value="placer_tool0_controller"/>
  </node>

  
  <node name="ur_driver_attacher" pkg="ur_modern_driver" type="ur_driver" output="screen">
  
    <param name="prefix" type="str" value="attacher_" />
    <param name="robot_ip_address" type="str" value="$(arg attacher_robot_ip)" />
    <param name="min_payload" type="double" value="$(arg min_payload)" />
    <param name="max_payload" type="double" value="$(arg max_payload)" />
    <param name="max_velocity" type="double" value="$(arg max_velocity)" />
    <param name="servoj_time" type="double" value="$(arg servoj_time)" />
    <param name="base_frame" type="str" value="attacher_base"/>
    <param name="tool_frame" type="str" value="attacher_tool0_controller"/>
  </node>

</launch>`

raequin, 2018-08-01:
**update:** Running a ur_modern_driver node on two computers sort of works.  MoveIt! starts okay via move_group.launch, but it continually prints these warnings:

[ WARN] [1533150373.142364481]: processStatus: Previously received status from [/ur_driver_placer], but we now received status from [/ur_driver_attacher]. Did the ActionServer change?
[ WARN] [1533150373.326288646]: processStatus: Previously received status from [/ur_driver_attacher], but we now received status from [/ur_driver_placer]. Did the ActionServer change?
[ WARN] [1533150373.340980873]: processStatus: Previously received status from [/ur_driver_placer], but we now received status from [/ur_driver_attacher]. Did the ActionServer change?

The big problem is that one of the robots will not move.  I can receive its position through moveit_commander but when I try to get it to move the ur_modern_driver node says

[ERROR] [1533150242.472278021]: Received a goal without velocities

It's only one of the robots that does this.  The other one works fine.  I've tried running the driver on different computers and the behavior is the same.  That is,

Manipulator 1 <-> Computer A
Manipulator 2 <-> Computer B

yields Manipulator 2 does not move.  So also does

Manipulator 1 <-> Computer B
Manipulator 2 <-> Computer A

In both of these instances the ROS master was running on Computer A.

raequin, 2018-08-01:
Ha ha ha!  The problem was that Manipulator 2 was in a pose from a different project where one of the joints was outside of the +/- pi range specified for the joints in the URDF.  It looks like two computers, each running a ur_modern_driver node works acceptably.  I would still appreciate it if you could tell me (A) your thoughts on one computer running two driver nodes, and (B) what to do about the warnings about receiving status from different nodes.

gavanderhoorn, 2018-08-01:
The only thing you'd need to change is the `reverse_port` parameter. Only a single process can bind port `50001`, so for two instances you'll have to make one of the nodes use a different port.

Pushing the two nodes in two different namespaces is something I would definitely recommend though. It'll make things a lot easier to manage.

raequin, 2018-08-01:
Thanks for the comments, @gavanderhoorn.  Here are three additional questions:

1. Is it appropriate to ask for guidance by creating an issue as I did here, or should I post such requests for help to answers.ros.org?

2. Do you think using group tags for the `ur_modern_driver` nodes would get rid of the warning about changing status sources?

3. Where can I change the `reverse_port` parameter?  (I don't see it in the launch files.)

gavanderhoorn, 2018-08-02:
> Is it appropriate to ask for guidance by creating an issue as I did here, or should I post such requests for help to answers.ros.org?

I would say answers is better, as we'd like to keep the issue tracker for actual problems, not configuration questions.

> Do you think using group tags for the `ur_modern_driver` nodes would get rid of the warning about changing status sources?

I'm not certain, I don't recall seeing that error.

Regardless I would suggest namespacing.

> Where can I change the `reverse_port` parameter?  (I don't see it in the launch files.)

It appears it's not listed in the launch files, but it's just a launch file `arg` and then `param` for the `ur_driver` node. See [this](https://github.com/ros-industrial/universal_robot/blob/2af9088dfa2671c8184900c0710e85ea54f2413f/ur_bringup/launch/ur10_bringup.launch#L13) for an example in the original `ur_driver` (ie: in `ros-industrial/universal_robot`).

You'll have to add it to the `urN_bringup.launch` you're using, and then to the `ur_common.launch`. Similar to how it's done in the example I linked.

gavanderhoorn, 2018-08-02:
As this is a configuration issue (at least it seems like one for now) I'm going to close this.

Do feel free to keep commenting on the issue btw.

If it turns out something is actually wrong with the code, we can re-open if that is appropriate.

closed 2018-08-02 by gavanderhoorn

Ashen1994, 2018-08-04:
Hello raequin, What does this sentence mean?（The problem was that Manipulator 2 was in a pose from a different project where one of the joints was outside of the +/- pi range specified for the joints in the URDF.）Can you descrbe it in detail? 
When I use the ur_modern_drive to control two arms, one of them was uncontrolled just like having a node giving it an unreachable position command. 
When I drive an arm alone, there is no problem with both arms.
Here is the launch file
<?xml version="1.0"?>

<launch>

  <arg name="ip_ur5" default="10.64.9.82"/>
  <arg name="ip_ur10" default="10.64.9.83"/>
  <arg name="limited_ur5" default="false"/>
  <arg name="limited_ur10" default="false"/>
  <arg name="min_payload_ur5"  default="0.0"/>
  <arg name="max_payload_ur5"  default="5.0"/>
  <arg name="min_payload_ur10"  default="0.0"/>
  <arg name="max_payload_ur10"  default="10.0"/>
  <arg name="prefix_ur5" default="ur5_" />
  <arg name="prefix_ur10" default="ur10_" />
  <arg name="controller_name_ur5" default="ur5"/>
  <arg name="controller_name_ur10" default="ur10"/>
  <arg name="max_velocity_ur5" default="10.0"/> 
  <arg name="max_velocity_ur10" default="10.0"/> 

  
  <include file="$(find dual_arm_robot_description)/launch/dual_arm_robot_upload.launch">
    <arg name="limited_ur5" value="$(arg limited_ur5)"/>
    <arg name="prefix_ur5" value="$(arg prefix_ur5)"/>
    <arg name="limited_ur10" value="$(arg limited_ur10)"/>
    <arg name="prefix_ur10" value="$(arg prefix_ur10)"/>
  </include>

  
  <group ns="$(arg controller_name_ur5)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="robot_ip_address" type="str" value="$(arg ip_ur5)"/>
      <param name="min_payload" type="double" value="$(arg min_payload_ur5)"/>
      <param name="max_payload" type="double" value="$(arg max_payload_ur5)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_ur5)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_ur5)" />
    </node>
    
    
    <rosparam file="$(find dual_arm_robot_driver)/config/ur5_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="ur5_joint_state_controller ur5_force_torque_sensor_controller ur5_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load ur5_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>

  
  <group ns="$(arg controller_name_ur10)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="reverse_port" type="int" value="5002"/>
      <param name="robot_ip_address" type="str" value="$(arg ip_ur10)"/>
      <param name="min_payload" type="double" value="$(arg min_payload_ur10)"/>
      <param name="max_payload" type="double" value="$(arg max_payload_ur10)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_ur10)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_ur10)" />
    </node>
    
    
    <rosparam file="$(find dual_arm_robot_driver)/config/ur10_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="ur10_joint_state_controller ur10_force_torque_sensor_controller ur10_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load ur10_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>

</launch>

Ashen1994, 2018-08-04:
Hello raequin, What does this sentence mean?（The problem was that Manipulator 2 was in a pose from a different project where one of the joints was outside of the +/- pi range specified for the joints in the URDF.）Can you descrbe it in detail?
When I use the ur_modern_drive to control two arms, one of them was uncontrolled just like having a node giving it an unreachable position command.
When I drive an arm alone, there is no problem with both arms.
Here is the launch file
<?xml version="1.0"?>

<launch>

  <arg name="ip_ur5" default="192.168.1.100"/>
  <arg name="ip_ur10" default="192.168.1.102"/>
  <arg name="limited_ur5" default="false"/>
  <arg name="limited_ur10" default="false"/>
  <arg name="min_payload_ur5"  default="0.0"/>
  <arg name="max_payload_ur5"  default="5.0"/>
  <arg name="min_payload_ur10"  default="0.0"/>
  <arg name="max_payload_ur10"  default="10.0"/>
  <arg name="prefix_ur5" default="ur5_" />
  <arg name="prefix_ur10" default="ur10_" />
  <arg name="controller_name_ur5" default="ur5"/>
  <arg name="controller_name_ur10" default="ur10"/>
  <arg name="max_velocity_ur5" default="10.0"/> 
  <arg name="max_velocity_ur10" default="10.0"/> 

  
  <include file="$(find dual_arm_robot_description)/launch/dual_arm_robot_upload.launch">
    <arg name="limited_ur5" value="$(arg limited_ur5)"/>
    <arg name="prefix_ur5" value="$(arg prefix_ur5)"/>
    <arg name="limited_ur10" value="$(arg limited_ur10)"/>
    <arg name="prefix_ur10" value="$(arg prefix_ur10)"/>
  </include>

  
  <group ns="$(arg controller_name_ur5)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="robot_ip_address" type="str" value="$(arg ip_ur5)"/>
      <param name="min_payload" type="double" value="$(arg min_payload_ur5)"/>
      <param name="max_payload" type="double" value="$(arg max_payload_ur5)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_ur5)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_ur5)" />
    </node>
    
    
    <rosparam file="$(find dual_arm_robot_driver)/config/ur5_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="ur5_joint_state_controller ur5_force_torque_sensor_controller ur5_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load ur5_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>

  
  <group ns="$(arg controller_name_ur10)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="reverse_port" type="int" value="5002"/>
      <param name="robot_ip_address" type="str" value="$(arg ip_ur10)"/>
      <param name="min_payload" type="double" value="$(arg min_payload_ur10)"/>
      <param name="max_payload" type="double" value="$(arg max_payload_ur10)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_ur10)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_ur10)" />
    </node>
    
    
    <rosparam file="$(find dual_arm_robot_driver)/config/ur10_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="ur10_joint_state_controller ur10_force_torque_sensor_controller ur10_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load ur10_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>

</launch>

Ashen1994, 2018-08-04:
Hello! what's the meaning of this setence(The problem was that Manipulator 2 was in a pose from a different project where one of the joints was outside of the +/- pi range specified for the joints in the URDF.)
When I use the ur_modern_drive to drive two arms, one of them  is uncontrolled just like having a node that is always sending it an unreachable location.

raequin, 2018-08-09:
@Ashen1994 From what I've read and seen, MoveIt! runs into trouble when computing trajectories with the UR if joint limits are +/- 2Pi.  Therefore I've used the joint-limited urdf.xacro, however, the ur10_bringup.launch file does not do this.  The problem I experienced was that when I started working with the robots, one of them had a joint position that exceeded Pi radians and so that resulted in no plan since it was outside of the joint limits.