ros-industrial-attic/ur_modern_driver #187: Speedj missing parameters

Ashen1994, 2018-08-03:
[image]

Display type error when using ur_modern_drive, Speedj missing parameters. The speedj on the needs exactly the same three parameters but given two dispalyed on the  teach pendant.

gavanderhoorn, 2018-08-03:
There is quite a history with UR, `speedj(..)`, how many arguments it really needs and those nrs (*and* the interpretation / semantics of the arguments) changing over time.

See #92 (and the linked issues there, such as #100) for some previous discussion.

If this has changed again, the driver will have to be updated.

But without more information (ie: your controller, version of polyscope, how you are using the driver, etc), we can't really say anything.

Ashen1994, 2018-08-03:
Thank you ,you are right.I have solved this problem by upgrated the version of polyscope. 
void UrRealtimeCommunication::setSpeed(double q0, double q1, double q2,
		double q3, double q4, double q5, double acc) {
	char cmd[1024];
	if( robot_state_->getVersion() >= 3.3 ) {
		sprintf(cmd,
				"speedj([%1.5f, %1.5f, %1.5f, %1.5f, %1.5f, %1.5f], %f, 0.008)\n",
				q0, q1, q2, q3, q4, q5, acc);
	}
	else if( robot_state_->getVersion() >= 3.1 ) {
		sprintf(cmd,
				"speedj([%1.5f, %1.5f, %1.5f, %1.5f, %1.5f, %1.5f], %f)\n",
				q0, q1, q2, q3, q4, q5, acc);
	}
	else {
		sprintf(cmd,
				"speedj([%1.5f, %1.5f, %1.5f, %1.5f, %1.5f, %1.5f], %f, 0.02)\n",
				q0, q1, q2, q3, q4, q5, acc);		
	}

gavanderhoorn, 2018-08-03:
I'm not sure what you did actually.

Could you give a bit more details?

Ashen1994, 2018-08-04:
Before I modified，the version of my polyscope is 3.1. The speedj has one more parameter when the version is greater than 3.3. I'm not sure, but I upgrate the version of polyscope to 3.3.  Everything is well.

gavanderhoorn, 2018-11-01:
It would be great if this could be submitted as a PR against `kinetic-devel`.

gavanderhoorn, 2019-11-23:
Please use `ur_robot_driver` with CB3 and e-series controllers.

closed 2019-11-23 by gavanderhoorn