ros-industrial-attic/ur_modern_driver #188: Multiple robots in one controller

Ashen1994, 2018-08-04:
I am sorry to ask another question, I have already seen it.
Controlling multiple (two) robots #186, but I think it's not the same as my problem. To control two URs from a single computer with ur_modern_driver. I created the modified launch file pasted below,  the error displayed is the same as shown in the picture.
<?xml version="1.0"?>

<launch>

  <arg name="ip_ur5" default="192.168.1.101"/>
  <arg name="ip_ur10" default="192.168.1.102"/>
  <arg name="limited_ur5" default="false"/>
  <arg name="limited_ur10" default="false"/>
  <arg name="min_payload_ur5"  default="0.0"/>
  <arg name="max_payload_ur5"  default="5.0"/>
  <arg name="min_payload_ur10"  default="0.0"/>
  <arg name="max_payload_ur10"  default="5.0"/>
  <arg name="prefix_ur5" default="ur5_" />
  <arg name="prefix_ur10" default="ur10_" />
  <arg name="controller_name_ur5" default="ur5"/>
  <arg name="controller_name_ur10" default="ur10"/>
  <arg name="max_velocity_ur5" default="10.0"/> 
  <arg name="max_velocity_ur10" default="10.0"/> 

  
  <include file="$(find dual_arm_robot_description)/launch/dual_arm_robot_upload.launch">
    <arg name="limited_ur5" value="$(arg limited_ur5)"/>
    <arg name="prefix_ur5" value="$(arg prefix_ur5)"/>
    <arg name="limited_ur10" value="$(arg limited_ur10)"/>
    <arg name="prefix_ur10" value="$(arg prefix_ur10)"/>
  </include>

  
  <group ns="$(arg controller_name_ur5)">
    <remap from="joint_states" to="/joint_states"/>
    
    <node name="ur_hardware_interface" pkg="ur_modern_driver" type="ur_driver" output="screen" launch-prefix="">
      <param name="robot_ip_address" type="str" value="$(arg ip_ur5)"/>
      <param name="min_payload" type="double" value="$(arg min_payload_ur5)"/>
      <param name="max_payload" type="double" value="$(arg max_payload_ur5)"/>
      <param name="max_velocity" type="double" value="$(arg max_velocity_ur5)"/>
      <param name="use_ros_control" type="bool" value="True"/>
      <param name="prefix" value="$(arg prefix_ur5)" />
    </node>
 
    <rosparam file="$(find dual_arm_robot_driver)/config/ur5_controllers.yaml" command="load"/>

    
    <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="ur5_joint_state_controller ur5_force_torque_sensor_controller ur5_vel_based_pos_traj_controller --shutdown-timeout 1" />

    
    <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load ur5_vel_based_admittance_traj_controller" />

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>

Ashen1994, 2018-08-04:
<?xml version="1.0"?>

<launch>

  <arg name="ip_ur5" default="192.168.1.101"/>
  <arg name="ip_ur10" default="192.168.1.102"/>
  <arg name="limited_ur5" default="false"/>
  <arg name="limited_ur10" default="false"/>
  <arg name="min_payload_ur5"  default="0.0"/>
  <arg name="max_payload_ur5"  default="5.0"/>
  <arg name="min_payload_ur10"  default="0.0"/>
  <arg name="max_payload_ur10"  default="5.0"/>
  <arg name="prefix_ur5" default="ur5_" />
  <arg name="prefix_ur10" default="ur10_" />
  <arg name="controller_name_ur5" default="ur5"/>
  <arg name="controller_name_ur10" default="ur10"/>
  <arg name="max_velocity_ur5" default="10.0"/> 
  <arg name="max_velocity_ur10" default="10.0"/> 

  
  <include file="$(find dual_arm_robot_description)/launch/dual_arm_robot_upload.launch">
    <arg name="limited_ur5" value="$(arg limited_ur5)"/>
    <arg name="prefix_ur5" value="$(arg prefix_ur5)"/>
    <arg name="limited_ur10" value="$(arg limited_ur10)"/>
    <arg name="prefix_ur10" value="$(arg prefix_ur10)"/>
  </include>

raequin, 2018-08-09:
Please let me know how you generated that graph.

gavanderhoorn, 2018-09-03:
>  what's the meaning of this setence
>> (The problem was that Manipulator 2 was in a pose from a different project where one of the joints was outside of the +/- pi range specified for the joints in the URDF.)

If configuring the URDF to load the "joint limited" version (ie: with all joints limited to `[-pi,pi]` instead of the regular `[-2pi,2pi]`) the physical robot must not be outside the `[-pi,pi]` range with any of its joints, or fi MoveIt will have difficulties working with the robot (as the joint(s) would appear to be outside the limits).

> When I use the ur_modern_drive to drive two arms, one of them is uncontrolled just like having a node that is always sending it an unreachable location.

That sounds like something that is unconnected to your first question.

If this is still a problem then please clarify what you're doing and perhaps we can help.

gavanderhoorn, 2018-09-03:
@raequin wrote:
> Please let me know how you generated that graph.

off-topic, but that is a screenshot of `rqt_graph`.

gavanderhoorn, 2018-11-01:
Closing this due to inactivity.

To give some closure to this: "multiple robots in one controller" would technically be impossible, as UR controllers cannot control multiple robots.

But running multiple instances of the ROS driver to communicate with multiple controllers is definitely possible: all that is needed is proper namespacing (as would be the case with any (set of) ROS node(s)) and setting the `reverse_port` to different values (as no two programs can listen on the same TCP port).

closed 2018-11-01 by gavanderhoorn