ros-industrial-attic/ur_modern_driver #189: The arm stops is joint speeds are not published at a high enough frequency.

parhartanvir, 2018-08-15:
Hi,
I am trying to control the arm motor speeds by publishing them to  `joint_speed`. If I publish to the topic with a  frequency of 250 Hz, in a while loop, it works fine.  But with the lower frequency, the arm jitters. Seems like if the arm resets the joint speeds to 0 if it does not receive a command for some time(which is really small). Is there a parameter that I can set to get around the same?
I can see  [here](https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_driver.cpp#L78), that the trajectory is interpolated between two consecutive points, for all points in the `JointTrajectory` message. Will I have to do something similar, or is it already implemented?

gavanderhoorn, 2018-08-22:
You don't tell us which particular model of UR robot you have, neither anything about your controller, so this is a bit of a generic reply.

---

> Seems like if the arm resets the joint speeds to 0 if it does not receive a command for some time(which is really small).

Velocity commands are cancelled after a nr of cycles have passed [here](https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_realtime_communication.cpp#L143-L146) (after being set [here](https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_realtime_communication.cpp#L98-L121)).

That nr of cycles is by default `12` ([here](https://github.com/ThomasTimm/ur_modern_driver/blob/9d202f450c0df467ec7e13f57bf38ee319dee05a/include/ur_modern_driver/ur_driver.h#L61-L62)). The cycle is coupled to the incoming data rate, which for the real-time connection is 125 Hz. I'm not sure why you need to publish at 250 Hz at the moment.

> Is there a parameter that I can set to get around the same?

The value is currently hard-coded to be `300` ([here](https://github.com/ThomasTimm/ur_modern_driver/blob/9397f47ac158fe5976719f33101334371bc15d44/src/ur_ros_wrapper.cpp#L90)) and cannot be configured via ROS parameters.

gavanderhoorn, 2018-11-01:
Closing this due to inactivity.

But I also don't think we can realistically do anything about this.

The current implementation of the driver sends new commands to the controller at a higher frequency than the internal command loop (or at least, on pre-e-series robots), so underruns of the velocity msg stream could cause this.

Something else to check would be the system software version and see whether it needs any special treatment with `speedj(..)`.

closed 2018-11-01 by gavanderhoorn