ros-industrial-attic/ur_modern_driver #193: How to publish movep via rosbridge and Json?

bastiannn, 2018-09-03:
I would like to publish movep commands from another software via rosbridge to the ur_modern_driver.
When using rosbridge with Json would it work like this code?
Is /ur_driver/URScript the correct topic and is the formatting of the std_msgs correct?

```
import json

data = {
"op":"publish",
"topic":/ur_driver/URScript,
"std_msgs/String":{
    "data":("movep(p[0.293,-0.514,0.125,0.2647,1.8632,-1.5809]), a=1.2, v=0.25)\n movep(p[0.293,-0.514,0.125,0.2647,1.8632,-1.5809]), a=1.2, v=0.25)\n
movep(p[0.293,-0.514,0.125,0.2647,1.8632,-1.5809]), a=1.2, v=0.25)\n")},
}

R = json.dumps(data)
```

gavanderhoorn, 2018-09-03:
I understand why you posted your question on this tracker, but just about everything you ask is about JSON and `rosbridge_suite`.

> Is /ur_driver/URScript the correct topic

yes.

>  is the formatting of the std_msgs correct?

It appears to be a `std_msgs/String`, so far so good.

As to the `data` field: I woule recommend to wrap all the lines in a program. Something like:

```python
def the_prog():
   # your lines
end
```
I'm not sure whether things will work otherwise.

gavanderhoorn, 2018-09-03:
As this is not an issue with the packages in this repository, I'm closing this.

If you feel this is incorrect please clarify and re-open. You may of course post additional comments regardless.

bastiannn, 2018-09-03:
Thanks for your fast reply!

I am trying to update the positions live from a CAD software, to live updated locations for poses. Which I than send to the robot.  So at the moment I am using a websocket connection from within my CAD software to send  data to topics. I can subscribe to topics of the ur_modern_driver, but have no idea how to send poses to it.

Why would it make more sense to not use the driver?
Or do you mean to send to another topic or even a service?

I am fairly new to ROS and this is my first attempt to operate an UR via ROS.

gavanderhoorn, 2018-09-03:
> Why would it make more sense to not use the driver?

you can just send URScript snippets/programs to `tcp://ip.of.your.robot:30002` directly. No need for the ROS driver.

bastiannn, 2018-09-03:
So I played a bit and now I am going for the topic  /follow_joint_trajectory/ goal
That seems to be more accurate for later tasks...

Anyhow I am getting the error received a goal with incorrect joint names.

`joint_names: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint',
               'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']`

Any ideas?

gavanderhoorn, 2018-09-03:
This seems like a completely new problem. It's generally recommended to not address different problems in a single issue.

While this sounds more like a ROS Answers question, if you post a new issue we can see whether we can address it here.