ros-industrial-attic/ur_modern_driver #194: Publishing to topic /follow_joint_trajectory/ goal

bastiannn, 2018-09-04:
I am trying to publish to the topic /follow_joint_trajectory/ goal
I am getting the error: 

> Received a goal with incorrect joint names.

`joint_names: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']`

To me the joint names look correct. Am I missing something else?

gavanderhoorn, 2018-09-04:
Have you created a composite xacro with the robot being only part of a scene and set the `prefix` argument?

If so: you'll have to update the names for the joints in your goal to match those with the prefix.

If not, then we're going to need more information before we can even begin to help you.

gavanderhoorn, 2018-09-04:
For future readers: this is a follow-up to #193.

bastiannn, 2018-09-05:
No, I did not add any prefix.
I guess I am doing something wrong in creating the trajectory_msgs/JointTrajectoryPoint Message.

Here the python code for creating the message:

`import json

topic = "goal"

data = {
  "op":"publish",
  "topic":topic,
"goal":{
    "trajectory":{
        "header":{
            "seq":(0),
            "stamp":(),
                "secs":(0),
                "nsecs":(0),
                "frame_id":("")
                },
        "joint_names":{
          "shoulder_pan_joint",
          "shoulder_lift_joint",
          "elbow_joint",
          "wrist_1_joint",
          "wrist_2_joint",
          "wrist_3_joint"
            },
        "points":{
            "positions:":[0.293,-0.514,0.125,0.2647,1.8632,-1.5809],
            "velocities":[0.0, 0.0, 0.0, 0.0, 0.0, 0.0],
            #"positions:":[0.2,-0.5,0.0,0.1,1.9,-1.7],
            #"velocities":[0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
            }
}
}
}
print(data)

R = json.dumps(data)

` 

It arrives in the topic though the fields joint_names and points stay empty.

gavanderhoorn, 2018-09-05:
`{}` is for maps in JSON, not lists. `joint_names` is a list, not a map.

Something similar is probably going on with `points`.

gavanderhoorn, 2018-09-06:
Seeing as you posted this on ROS Answers ([publishing to trajectory_msgs/JointTrajectoryPoint](https://answers.ros.org/question/302610)), I'm closing this one.

closed 2018-09-06 by gavanderhoorn