ros-industrial-attic/ur_modern_driver #195: UR5 arm is protective stopped during execution without touching any obstacles

shinian123, 2018-09-05:
Our lab use ROS operating system to operate UR5 arms using moveit. Recently,we encounterd a problem when the UR5 arms get protective stop during execution. While the arm does not touch any obstacles.

Our joint_limits.yaml is :
joint_limits:
left_arm_elbow_joint:
has_velocity_limits: true
max_velocity: 3.15
has_acceleration_limits: false
max_acceleration: 0
left_arm_shoulder_lift_joint:
has_velocity_limits: true
max_velocity: 3.15
has_acceleration_limits: false
max_acceleration: 0
left_arm_shoulder_pan_joint:
has_velocity_limits: true
max_velocity: 3.15
has_acceleration_limits: false
max_acceleration: 0
left_arm_wrist_1_joint:
has_velocity_limits: true
max_velocity: 3.2
has_acceleration_limits: false
max_acceleration: 0
left_arm_wrist_2_joint:
has_velocity_limits: true
max_velocity: 3.2
has_acceleration_limits: false
max_acceleration: 0
left_arm_wrist_3_joint:
has_velocity_limits: true
max_velocity: 3.2
has_acceleration_limits: false
max_acceleration: 0
right_arm_elbow_joint:
has_velocity_limits: true
max_velocity: 3.15
has_acceleration_limits: false
max_acceleration: 0
right_arm_shoulder_lift_joint:
has_velocity_limits: true
max_velocity: 3.15
has_acceleration_limits: false
max_acceleration: 0
right_arm_shoulder_pan_joint:
has_velocity_limits: true
max_velocity: 3.15
has_acceleration_limits: false
max_acceleration: 0
right_arm_wrist_1_joint:
has_velocity_limits: true
max_velocity: 3.2
has_acceleration_limits: false
max_acceleration: 0
right_arm_wrist_2_joint:
has_velocity_limits: true
max_velocity: 3.2
has_acceleration_limits: false
max_acceleration: 0
right_arm_wrist_3_joint:
has_velocity_limits: true
max_velocity: 3.2
has_acceleration_limits: false
max_acceleration: 0

It seems when the speed of the arms changed drastically , the protective stop program will be triggered.
We think the problem is due to some parameters of joint_limits.yaml is not correctly set, but we don't know how to tune these parameters.Could any one help us? Thanks.

cjsavoia, 2018-09-05:
Try to change the safety settings on the Teach pendant. These settings are
responsible for protective stops. Also if you are picking large objects
without changing the payload it will protective stop.

shinian123, 2018-09-05:
Thanks.But can we only change safe settings through teach pendant? Meanwhile, the arm can be protective stopped even we don't pick large objects.

shinian123, 2018-09-05:
@cjsavoia Thanks.But can we only change safe settings through teach pendant? Meanwhile, the arm can be protective stopped even we don't pick large objects.

cjsavoia, 2018-09-05:
Yes safety settings get changed on TP only. They are applied at startup of
arm.

gavanderhoorn, 2018-11-01:
@shinian123: are you still having issues with this?

Can you please clarify which version of the driver you are using? The `master` branch or the `kinetic-devel` branch?

henningkayser, 2018-11-09:
Hi, we also get protective stops in our UR5 setup using ROS and MoveIt.
We use the current version of the `kinetic-devel` branch.
In particular we get the following error messages:

* C153A0: Position Deviates from path: Base
* C204A3: Invalid setpoints: Sudden stop

We checked the safety settings and couldn't find anything that could be related to that behavior.
Our payload is already configured correctly.

Does anybody have an idea what could cause these issues?

Eric0751, 2018-12-01:
Hi, I encountered this problem a few days ago. I solve this by launching ur5_bringup.launch with "use_lowbandwidth_trajectory_follower:=true". I don't know what does this option mean but it does solve the Protective Stop problem.

I'm using ubuntu16.04 with ROS kinetic and ur_modern_driver of Zagitta/ur_modern_driver.[https://github.com/Zagitta/ur_modern_driver](url)

gavanderhoorn, 2019-11-23:
While `ur_robot_driver` does not support the "low bandwidth" mode, it does contain mitigations for dropped packages that should address the issues reported by the OP.

If similar issues do still occur, please report it over at the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn