ros-industrial-attic/ur_modern_driver #196: Can use this with robot movement interface package?

lakshmip001, 2018-09-06:
For my application , I need to record tool velocity while the ur_driver in the robot movement interface doesnt have topic tool_velocity. Can you please suggest how I can record velocity?

gavanderhoorn, 2018-09-06:
If you don't command any motion with `ur_modern_driver` then in theory you could start it and it would publish tool velocity without interfering with anything else running on the controller.

I've not used "robot movement interface" though, so I cannot guarantee it will work.

gavanderhoorn, 2018-09-06:
I'm going to close this as it's not really an issue, but a question about configuration / usage.

@lakshmip001: feel free to keep commenting though.

closed 2018-09-06 by gavanderhoorn