ros-industrial-attic/ur_modern_driver #197: UR robot dual arm manipulation

houlei516, 2018-09-09:
Hope to realize dual arm manipulation by two UR 10 robot with ur_modern_drive. Each UR arm has a different task but coordinates with each other, rather than master-slave. Already refer to https://answers.ros.org/question/269074/dual-arm-ur5-connection-with-universal-robot-stack/ 

Look forward to a generic method. Thanks.

gavanderhoorn, 2018-09-15:
There isn't anything really special about `ur_modern_driver`, and the same approach can be taken as with the old `ur_driver`: run two instances in two different namespaces and make sure to configure the `reverse_port` parameter with a unique value for each instance (it is the TCP port on which the driver will communicate with the controller, so two instances of the driver cannot share the same port).

At this point you'd have two copies of all topics, services and action servers.

If you have nodes that would benefit from a "global `JointState`" message, use `joint_state_publisher` together with the `source_list` parameter to arrange that.

gavanderhoorn, 2018-09-15:
I'm going to close this as it's a question about configuration of the node, not an actual issue.

You are free to keep commenting of course.

closed 2018-09-15 by gavanderhoorn

houlei516, 2018-09-16:
Someone suggests using the 'Moveit! Setup Assistant' to configure the other UR10 to avoid the namespace issue, as if they are a PR2 and alike. If you disagree, could you please teach me a systematic method to run two UR10 independently on one PC? Thank you. Our goal is to use one UR10 as a turntable, while the other UR10 operates on it, scanning, picking and etc. 

Thank you. I look forward to your help.

gavanderhoorn, 2018-09-16:
> Someone suggests using the 'Moveit! Setup Assistant' to configure the other UR10 to avoid the namespace issue, as if they are a PR2 and alike.

I don't know what this means.

houlei516, 2018-09-16:
A software engineer suggests using the Moveit! Setup Assistant http://docs.ros.org/hydro/api/moveit_setup_assistant/html/doc/tutorial.html 
to configure two ur10 into a dual arm system, from scratch.

gavanderhoorn, 2018-09-16:
That might be a good idea in any case, especially if you want to plan for both arms in a single scene.

That is almost orthogonal of how you configure the driver though.

> from scratch.

If "from scratch" means: following the wizard to create a new MoveIt configuration pkg, then I would recommend to *always* do that. 

The MoveIt pkgs we provide here are almost just examples. It's very rare that you would have your robot in isolation in free space, so a rerun of the MSA is almost always needed.

gavanderhoorn, 2018-09-16:
Btw: I'd recommend to use the latest version of the MSA documentation / tutorials: [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html).

The Hydro version you link is rather antiquated by now.

houlei516, 2018-09-16:
Thank you and happy weekend. 
I am using Kinect. Sorry for the casual google results. 

I will appreciate it if you can teach me your method for dual arm. I tried to modify the mentioned files in my first post. But it broke the ur_modern_drive. I cannot use Moveit! however purging and reinstalling the package. I reinstalled my laptop. Could you please teach me a systematic work flow to make two UR10 cooperate, in your understanding. Thank you very much. I can try to carry it out step by step. 

PS: Ubuntu 16 now uses the kernel 4.15. The reinstallation is the cause of failing to build the realsense2 package.