ros-industrial-attic/ur_modern_driver #204: Current trajectory execution fails with protective stop after calling service to set IO pin

sharathrjtr, 2018-10-03:
I've a Cartesian path motion plan. This is executed with the group.asyncExecute(cartesian_plan). During the execution of trajectory the IO pin 0 is made as digital output and set HIGH by service call to /ur_driver/set_io. But the trajectory execution fails with protective stop as soon as the service is called. 
Note: There is no error if i call set_io pin before or after the execution of the trajectory.
Here the output from the terminal. I've called the service twice.
```
[ INFO] [1538547367.747234543]: Received new goal
[ INFO] [1538547367.747371174]: Trajectory received and accepted
[ INFO] [1538547367.747469439]: Translating trajectory
[ INFO] [1538547367.747538490]: Executing trajectory with 10 points and duration of 3.395s
[ INFO] [1538547367.747583287]: Uploading trajectory program to robot
[ INFO] [1538547371.031811875]: setIO called with [1, 0]
[ INFO] [1538547371.035902120]: setIO called with [1, 0]
[ INFO] [1538547371.042680691]: Trajectory failed
```

gavanderhoorn, 2018-10-03:
If you are using the `master` branch then this is expected: IOs are set using a regular *primary* program, causing the regular trajectory executing program to be stopped in its tracks.

The version in `kinetic-devel` uses *secondary* programs to interact with IO (added in https://github.com/Zagitta/ur_modern_driver/pull/21). That should allow it to be done concurrently with motion.

I haven't yet announced the `kinetic-devel` version, but be sure to specify `CMAKE_BUILD_TYPE=RelWithDebInfo` when building it, as it benefits from compiler optimisations quite a bit.

sharathrjtr, 2018-10-03:
Thanks a lot!! It working now.

closed 2018-10-03 by sharathrjtr