ros-industrial-attic/ur_modern_driver #205: roslaunch is not working

BigHopes, 2018-10-03:
Hi all,

please help me with this problem, I have been trying many things but no success:

I have ubuntu 16.04 and ROS kinetic (downloaded through [steps](http://wiki.ros.org/kinetic/Installation/Ubuntu)). I created the catkin workspace ([link](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment)). 

Then in the catkin_ws/src I did: 
git clone https://github.com/ros-industrial/ur_modern_driver.git 

but then when I made catkin_make:
it gave this final error: "Invoking "make -j2 -l2" failed"
when I read the errors before it said:  
"error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’"

I tried to complete running the ur by: roscore (it works correct)
and "roslaunch ur_modern_driver ur5_bringup.launch robot_ip:="
it gives this error:
"ERROR: cannot launch node of type [ur_modern_driver/ur_driver]: can't locate node [ur_driver] in package [ur_modern_driver]"

What should I do?
Any help is appreciated.

Thanks in advance

gavanderhoorn, 2018-10-03:
The version in the `master` branch is not compatible with Kinetic. You'll have to check out the version in `kinetic-devel`.

Also: there is no point in trying to `roslaunch` anything if the package didn't build correctly. Use of the node in this package will depend on successful compilation, so without that, `roslaunch` will not be able to launch anything. That is not specific to this package, but is true for all ROS packages and nodes.

gavanderhoorn, 2018-10-03:
I'm going to close this as it's not an issue with the package itself.

You are free to keep commenting on this issue of course.

closed 2018-10-03 by gavanderhoorn

BigHopes, 2018-10-03:
I am new in ROS, please can you tell me how to check the version?

gavanderhoorn, 2018-10-03:
It's not the ROS version, but the version of `ur_modern_driver` that you are compiling.

By running `git clone ..`, you checkout the `master` branch.

You'll need to change that so that `git` checks out the `kinetic-devel` branch.

If you're not familiar with `git`, I would encourage you to follow a tutorial or two about it. You'll benefit greatly from some experience with it when working with ROS.

In this particular case, you could do something like this:

```
cd /path/to/your/workspace/src/ur_modern_driver
git checkout kinetic-devel
```

At this point clean your workspace and try rebuilding it.

gavanderhoorn, 2018-10-03:
I'm assuming here that you've followed the regular procedure for building a package from sources. That is: you've made sure to install all dependencies of the package first.

If you haven't, I would suggest to take a look at [question 252478](https://answers.ros.org/question/252478) on ROS Answers.

BigHopes, 2018-10-03:
Now the catkin_make worked :) Thanks

but roslaunch is not working again with the same error "ERROR: cannot launch node of type [ur_modern_driver/ur_driver]: can't locate node [ur_driver] in package [ur_modern_driver]""

BigHopes, 2018-10-03:
btw I did it like: git checkout -b kinetic-devel (cause the previous one didn't work)

BigHopes, 2018-10-03:
I believe the problem is because catkin_ws/devel/lib is empty,
it should have ur_driver in it right? how can I do that?

gonzalocasas, 2018-10-03:
@BigHopes silly question but did you source your workspace after building? (i.e. `source ~/catkin_ws/devel/setup.bash`)

BigHopes, 2018-10-03:
sure I did 
Also, I add it to the file: ~/.bashrc

gavanderhoorn, 2018-10-05:
Please tell us *exactly* what you are doing: so all commands, in order, in a comment here.

Don't just point to tutorials. I need to know the exact commands you used, otherwise I cannot help you.

BigHopes, 2018-10-05:
Sure, there are the detailed steps (using ROS kinetic and ubuntu 16.04, I can write the same steps in ROS indigo too and ubuntu 14.04 and same error):
---------------
1- download ubuntu 16.04
http://releases.ubuntu.com/16.04.5/ (ubuntu-16.04.5-desktop-amd64.iso )
--------------
2- install ROS kinetic
http://wiki.ros.org/kinetic/Installation/Ubuntu (all the steps from 1.1 to 1.7, using Desktop-Full Install version)
---------------
3- source: $ source /opt/ros/kinetic/setup.bash
---------------
4- create catkin_ws ([src](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment)), as:

$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/
$ catkin_make

$ source devel/setup.bash

$ echo $ROS_PACKAGE_PATH
and it gives: /home/youruser/catkin_ws/src:/opt/ros/kinetic/share
---------------
5- then in the catkin_ws/src directory:
git clone https://github.com/ros-industrial/ur_modern_driver.git
---------------
6- then in catkin_ws I wrote: catkin_make
---------------
7- then in new terminal: roscore
---------------
8- in another new terminal (pc ip: 10.1.1.5, robot ip: 10.1.1.4): 
roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=10.1.1.4

and here i got the error:
"ERROR: cannot launch node of type [ur_modern_driver/ur_driver]: can't locate node [ur_driver] in package [ur_modern_driver]"
---------------------------------------------
Thanks in advance

miguelprada, 2018-10-05:
Also, if you follow the steps in your last comment, @BigHopes, the `catkin_make` in step 6 must've failed. You should switch to `kinetic-devel` branch in ur_modern_driver for the driver to build succesfully against ROS kinetic.

TL;DR;

5.5 Switch to kinetic-devel branch:
```
cd ~/catkin_ws/src/ur_modern_driver
git checkout -t origin/kinetic-devel
```

BigHopes, 2018-10-05:
@gavanderhoorn I did that and the `rospack find ur_modern_driver` print:
catkin_ws/src/ur_modern_driver

@miguelprada I did that before but by `git checkout -b kinetic-devel` (as @gavanderhoorn said [here](https://github.com/ros-industrial/ur_modern_driver/issues/205#issuecomment-426625249)) and then `catkin_make` works but roslaunch gave the same error about package.

I tried `git checkout -t origin/kinetic-devel` and then `catkin_make` but gives this error:
`couldn't find a package configuration file provided by "insdustrial_msgs"`

gavanderhoorn, 2018-10-05:
So `catkin_make` cannot have been successful, as it would appear you didn't have all dependencies.

This would be the procedure to build everything successfully:

```
source /opt/ros/kinetic/setup.bash
mkdir -p $HOME/catkin_ws/src
cd $HOME/catkin_ws
git -C src/ clone -b kinetic-devel https://github.com/ros-industrial/ur_modern_driver.git
rosdep update
rosdep install --from-paths src/ --ignore-src
catkin_make
source $HOME/devel/setup.bash
```

At this point `catkin_make` should not have shown you any errors, `rospack find ur_modern_driver` should return the correct path and a `roslaunch ur_modern_driver ..` should succeed.

Please make sure to check each step for errors.

The fact that you get the command line prompt back does not mean things worked.

BigHopes, 2018-10-05:
The robot moves yeeeeh :) :) :) 
Finally :) Thank you for everything.

I will edit the steps for others.

gavanderhoorn, 2018-10-05:
Good to hear you got it to work.

Note btw that these steps are the default ones you would always use when building ROS packages from sources.

BigHopes, 2018-10-05:
Ok, will write them just in case somebody stuck in that like me :)

BigHopes, 2018-10-05:
The detailed steps (using ROS kinetic and ubuntu 16.04, or ROS indigo and ubuntu 14.04):
(it may be obvious but it was a problem for me)

1- download ubuntu 16.04 (or ubuntu 14.04)
http://releases.ubuntu.com/16.04.5/ (ubuntu-16.04.5-desktop-amd64.iso )

2- install ROS kinetic (or ROS indigo)
http://wiki.ros.org/kinetic/Installation/Ubuntu (all the steps from 1.1 to 1.7, using Desktop-Full Install version)

3- source: $ source /opt/ros/kinetic/setup.bash

4- create catkin_ws (src), as:

```
$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/
$ catkin_make
```

`$ source devel/setup.bash`

`$ echo $ROS_PACKAGE_PATH`
should gives: /home/youruser/catkin_ws/src:/opt/ros/kinetic/share

5- write those:

```
source /opt/ros/kinetic/setup.bash
mkdir -p $HOME/catkin_ws/src
cd $HOME/catkin_ws
git -C src/ clone -b kinetic-devel https://github.com/ros-industrial/ur_modern_driver.git
rosdep update
rosdep install --from-paths src/ --ignore-src
catkin_make
source $HOME/devel/setup.bash
```

5- then in new terminal: `roscore`

6- in another new terminal (pc ip: 10.1.1.5, robot ip: 10.1.1.4):
`source catkin_ws/devel/setup.bash`
`roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=10.1.1.4`

6- in another new terminal:
`source devel/setup.bash`
`rosrun ur_modern_driver test_move.py`
...........................
note: in every new terminal you should 'source devel/setup.bash'
or you can add that to a file:
`sudo nano  ~/.bashrc`
go to the end of it, you will find this line:
`source /opt/ros/indigo/setup.bash`
after it you should add this line:
`source ~/catkin_ws/devel/setup.bash`

press ctrl+o to save and press enter
then ctrl+x to exit

close the terminal then open it again
...........................

Thanks for @gavanderhoorn  and @miguelprada  for their help.

Anastasiamor100, 2019-02-05:
When following your procedure (step 5) on a melodic system, after entering the following command I got an error: 
`rosdep install --from-paths src/ --ignore-src`

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_modern_driver: Cannot locate rosdep definition for [industrial_msgs]`

Do you know how to fix this problem?

gavanderhoorn, 2019-02-05:
Before your edit this was a completely different problem.

Could I please ask you to post a new issue, as after your edit, this is no longer related to the issue that you posted on.