ros-industrial-attic/ur_modern_driver #208: ROS UR5 any working steps?

BigHopes, 2018-10-05:
Please help me in installing the driver, I was trying for about 5 days with no result.
I tried with ROS kinetic (on ubuntu 16.04)
also with ROS indigo (on ubuntu 14.04)

I followed the ROS downloading steps ([kinetic](http://wiki.ros.org/kinetic/Installation/Ubuntu), [indigo](http://wiki.ros.org/indigo/Installation/Ubuntu)). Then I cloned the ur_modern_drive to catkin_ws/src and made catkin_ws (it works) then rosrun.

When I call:
"roslaunch ur_modern_driver ur5_bringup.launch robot_ip:="
I get the following error:
"ERROR: cannot launch node of type [ur_modern_driver/ur_driver]: can't locate node [ur_driver] in package [ur_modern_driver]"

Should I use kinetic or indigo? Are there any extra steps that I should follow?
Please help me, ready to try any new thing.

gavanderhoorn, 2018-10-05:
Please don't open new issues for the same problem.

Let's continue this in #205.

closed 2018-10-05 by gavanderhoorn

BigHopes, 2018-10-05:
Can you still help me with that issue? 
I didn't want to disturb you that is why I opened another issue.

gavanderhoorn, 2018-10-05:
From what I can see the issue is exactly the same.

I'm closing this one so we can continue in #205.

Please describe what is happening in #205.