ros-industrial-attic/ur_modern_driver #217: hardware_interface::RobotHW & custom controllers

v4hn, 2018-10-22:
In personal conversation @bmagyar noticed that this driver does not "properly" implement `ros_control`, and I agree.

To the best of my knowledge, the driver can not be used to form a `combined_robot_hw::CombinedRobotHW` (as it does not provide a `RobotHW` plugin) and it cannot be used with custom controllers (the `load_controllers` plugin api is stubbed).

Talking to engineers at Shadow Robotics, esp. @toliver, I found they wrote [*yet another* UR driver](https://github.com/shadow-robot/sr_ur_arm/tree/kinetic-devel/sr_ur_robot_hw) (horray), that implements ros_control better.

I believe their implementation is quite superior w.r.t. `ros_control` and this package should support this the same (or similar) way.

v4hn, 2018-10-23:
> but I'm not sure I understand what you mean with the second. What do you mean with custom controllers in this context?

I withdraw this claim. It has been a while since I last looked into this code base.
I seemed to recall that it was impossible to load custom `ros_control` plugins with the implementation, but I can't find anything in the repository at the moment that would indicate that.

The main motivation for this issue was to point out Shadow Robotic's UR driver after they told me about it in person.
I wanted to give them a gentle push to collaborate with the community here, instead of fixing every bug they find with every new polyscope release in-house. :)

> Also, is SR's driver better or superior in any other way w.r.t. ros_control? Or is it just regarding these two aspects?

I did not use SR's driver myself. I talked to them at their IROS booth where they tested their driver with a more current Polyscope version.
@toliver could you or one of the others elaborate on the features supported by your driver?

miguelprada, 2018-10-23:
It is definitely possible to load custom controller plugins, yes.

In any case, thanks for the information, @v4hn. The possibility to compose multiple `RobotHW` instances into a single `ros_control` node (e.g. for dual arm configurations) is very interesting and I will certainly look into it.

gavanderhoorn, 2018-11-01:
I'm all for improving the `hardware_interface` and make it compatible with `CombinedRobotHW`, but that might need some planning so as to not overlap with the developments going on in some potential ROSIN FTPs.

toliver, 2019-02-07:
I apologise for being so late to this discussion.

I don't know in depth what are advantages or the performance of the ur_modern_driver as we still haven't tested it. But after a quick check to the readme and the code I think that the only extra feature that Shadow Robot's driver is adding is compatibility with CombinedRobotHW.

`sr_ur_arm` features:
- control of the arm through servoj commands (like in the ros_control mode of `ur_modern_driver`)
- service to set arm speed (percentage speed seen in the teach pendant)
- service to set end effector mass and centre of mass

Disadvantages of `sr_ur_arm` vs  `ur_modern_driver`:
- not able to send speed commands (not implemented yet)
- currently we are not providing a mechanism to synchronize the state readings from the arm with the controller update and the command sending (like the modern driver provides [here](https://github.com/ros-industrial/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L584)).

One of the unsolved limitations of CombinedRobotHW is how to deal with multiple robots that have different update rates. In our case we use the Shadow hand (1000Hz) and UR10 (125Hz).
The way we are currently dealing with it is by running [a 1000Hz control loop](https://github.com/shadow-robot/ros_control_robot/blob/kinetic-devel/src/main.cpp#L323-L326) and only running [every 8th call to read() and write()](https://github.com/shadow-robot/sr_ur_arm/blob/kinetic-devel/sr_ur_robot_hw/src/sr_ur_robot_hw.cpp#L169) for the UR RobotHW. This is currently hardcoded but it's easy to make it a parameter.

The mentioned approach is updating the arm controllers at 1000Hz but only sending servoj commands to the arm at 125Hz. This seems to work well with joint_trajectory_controllers and servoj, but I'm not sure if it would work with trajectories using speedj.

Currently we don't have a short term plan to work on UR drivers, but maybe in the following months we will test the  `ur_modern_driver` and if we see an advantage to using it (e.g. if speedj improves the behaviour of the arm for our use cases) we might contribute to make it compatible with CombinedRobotHW.

gavanderhoorn, 2019-11-23:
There might still be interesting features in `sr_ur_arm` that could be ported to / integrated with `ur_robot_driver`.

Initial work towards supporting `CombinedRobotHW` has been merged in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/34.

closed 2019-11-23 by gavanderhoorn

toliver, 2019-11-25:
We will do some tests using the new driver as a plugin with `CombinedRobotHW`.
So far the only issues that I foresee have to do with the timing of the loop, that in the new driver is driven by the state frames coming from the robot. We might have to change that so that it is not blocking.

gavanderhoorn, 2019-11-25:
> So far the only issues that I foresee have to do with the timing of the loop, that in the new driver is driven by the state frames coming from the robot. We might have to change that so that it is not blocking.

yes, this has also been discussed in the referenced PR.

To be honest I actually feel having the robot controller "in control" of the control flow on the `ros_control` side makes a lot of sense to me.

I realise that complicates combined_hw integration though. Perhaps a good approach can be found to nicely work around/with it though.