ros-industrial-attic/ur_modern_driver #219: Real time communication with simple action client issue

JuiceGrape, 2018-10-23:
Hey there,

I've been using this for a project, moving a ur10 in real time using an HTC Vive tracker. However, I have run into a problem. 
I'm using a simple action client to send FollowJointTrajectoryGoals to the robot, and it's moving fine. However, if I start sending small movements, it simply refuses to move, giving me a C204A3 path sanity check error, saying the following:
[image]

Any ideas or is it a bug that I will have to work around?

AndyZe, 2018-10-23:
There are several ways. I'm working through similar issues. My advice would be to avoid the kinetic-devel branch for now. I've used this fork with success on Kinetic: https://github.com/iron-ox/ur_modern_driver

If using the iron-ox fork, you can publish a `trajectory_msgs/JointTrajectory` to the topic /joint_speed. You could also stream a text string to the ur_script topic (search ROS answers for how to do that). If you use these methods, there will not be collision checking or other error handling unless you implement it yourself.

Better yet, use the [jog_arm](https://github.com/UTNuclearRoboticsPublic/jog_arm) package. Then you will be able to specify whatever coordinate frame you want and there will be collision checking, joint limit checking, and singularity avoidance. It's actually fairly easy to use and I can walk you through the setup, if needed.

AndyZe, 2018-10-24:
It does use MoveIt! for some tasks like loading the robot model and collision checking, but it sends its own command stream to the robot/simulation. The steps to use it are:

1. clone and build the iron-ox fork of ur_modern_driver and switch to the iron-kinetic-devel branch
2. launch the MoveIt! stuff like normal:

  -roslaunch ur_modern_driver ur3_bringup.launch robot_ip:=192.168.1.102
  -roslaunch ur3_moveit_config ur3_moveit_planning_execution.launch
  -roslaunch ur3_moveit_config moveit_rviz.launch config:=true

3. put the attached launch and config files somewhere you can find them
4. launch the jog_arm launch file

  - roslaunch ~/Desktop/my/path/jog_node.launch

5. you can change the command frame and speed, etc. in the config file

Now your VR setup just needs to publish a **geometry_msgs/TwistStamped** to the **/jog_arm_server/delta_jog_cmds** topic. The message should look like this:

> header:
>   seq: 5909
>   stamp:
>     secs: 1540390430
>     nsecs: 844682101
>   frame_id: 'base_link'
> twist:
>   linear:
>     x: 0.8
>     y: 0.02
>     z: 0.0
>   angular:
>     x: 0.0
>     y: 0.0
>     z: 0.0

(Note, all elements of the message should be between -1:1 or you will get a warning.)

[jog_node.launch.txt](https://github.com/ros-industrial/ur_modern_driver/files/2510936/jog_node.launch.txt)

[jog_settings.yaml.txt](https://github.com/ros-industrial/ur_modern_driver/files/2510939/jog_settings.yaml.txt)

JuiceGrape, 2018-10-25:
Getting this error: 

process[jog_arm_server-1]: started with pid [15503]
[ INFO] [1540475088.112051507]: Loading robot model 'ur10'...
[ WARN] [1540475088.112188986]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1540475088.112215936]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1540475088.196693011]: Loading robot model 'ur10'...
[ WARN] [1540475088.196728137]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1540475088.196743977]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1540475088.241065678]: Loading robot model 'ur10'...
[ WARN] [1540475088.241097007]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1540475088.241110913]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1540475088.244281789]: Waiting for first joint msg.
[ INFO] [1540475088.304312604]: Loading robot model 'ur10'...
[ WARN] [1540475088.304374340]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1540475088.304397533]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1540475088.325622974]: Group 'arm' was not found.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Group 'arm' was not found.

Any ideas?
I'm running a ur10 btw, so all of the commands were changed to match

AndyZe, 2018-10-25:
Doesn't the move_group usually default to 'manipulator' on UR robots?

If RViz is running, check by going to Motion Planning -> Planning Request -> Planning Group.

JuiceGrape, 2018-10-25:
It is indeed set to manipulator

AndyZe, 2018-10-25:
OK, good. In your jog_settings.yaml file, is the **move_group_name** set to **manipulator**, too? That's all I can think of that would cause this error.

JuiceGrape, 2018-10-25:
It is set to that yes

AndyZe, 2018-10-25:
Maybe the correct yaml file isn't being loaded. Can you check that the launch file is pointing toward the right yaml file.

My guess is, it's still set to the example that comes with the package:
<rosparam command="load" file="$(find jog_arm)/config/jog_settings.yaml" />

gavanderhoorn, 2018-10-26:
As this has gone largely off-topic for this list, I'm closing this.

As to the original question and issue: the driver does not support trajectory replacement at this time, so implementing jogging or incremental motion using many small trajectories is not something that is possible. The controller entering protective stops could very well be due to your trajectories not being continuous in acceleration or jerk, leading to situations where the UR controller decides that it's not a well-formed trajectory and the robot should be stopped.

closed 2018-10-26 by gavanderhoorn