ros-industrial-attic/ur_modern_driver #220: kinetic-devel branch: erratic motion with ur3_ros_control launch file

AndyZe, 2018-10-23:
I find that both the` /vel_based_pos_traj_controller` and the` /pos_based_traj_controller` produce erratic motions for the UR3. The velocity-based controller is highly unstable and takes off immediately ([video](https://drive.google.com/file/d/1Temqsfckh5fECnJuTUH8lZWd8mo3p2QJ/view?usp=sharing)) while the position-based controller oscillates with large magnitude about the setpoint ([video](https://drive.google.com/file/d/1H97_EFh-2xF-uiryyOl_Tz2FMlLV_nMH/view?usp=sharing)). The PID gains on the velocity-based controller are ridiculously high (Kp=50) but even if I turn them down to Kp=0.01, Kd=0.001, the robot is unstable. I'm not sure how to go about tuning the position-based controller because ros_control does not expose PID gains for it.

I launch with `roslaunch ur_modern_driver ur3_ros_control.launch robot_ip:=192.168.1.102`

and I added the appropriate controller to `ur3_moveit_config/config/controllers.yaml` like this:

```
controller_list:
 - name: /vel_based_pos_traj_controller #or /pos_based_pos_traj_controller
   action_ns: follow_joint_trajectory
   type: FollowJointTrajectory
   default: true
   joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
```

My UR3 software is version 3.5.0.1.10584, ur_modern_driver commit c4b2c73, universal_robot commit f528e20 (a.k.a. the most recent).

miguelprada, 2018-10-23:
The velocity controller video looks like you may be hitting #206 more than an oscillation going unstable. Could you try with the fix proposed in #213 and see if that solves the problem for you?

Regarding the position controller, no PID gains are exposed because there's no PID controller involved at all. I'm truly baffled that you are experiencing that oscillatory behavior. Are you absolutely sure its the `pos_based_pos_traj_controller` you are running?

AndyZe, 2018-10-23:
OK, I checked that the pos_based_pos_traj_controller was actually loaded. This time, it took off similar to how the vel_based_pos_traj_controller did.

I'm looking forward to testing your PR.

AndyZe, 2018-10-24:
@miguelprada Your fix works well for me. I made a PR to set the PID gains for the UR3 velocity-based position controller to reasonable values.

closed 2018-10-24 by AndyZe