ros-industrial-attic/ur_modern_driver #221: Open and run a program

martamillet, 2018-10-24:
Hi,

I wanted to ask if it's possible from ros to open and run a progam that is locally saved in the controller of the ur?

Thanks!!

gavanderhoorn, 2018-10-25:
No, that is currently not directly supported (as in: there is no ROS API dedicated to that), although you could use the `URScript` topic for that.

gavanderhoorn, 2018-10-25:
I'm closing this as it's not something actionable, but a question about functionality offered.

You're free to keep commenting on this issue of course.

closed 2018-10-25 by gavanderhoorn

miguelprada, 2018-10-25:
In addition to being able to send the program you with to the `URScript` topic as @gavanderhoorn suggested, the [dashboard server](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/dashboard-server-cb-series-port-29999-15690/) also offers the possibility to load/start programs saved inside the controller itself. However, I think I haven't been able to succesfully use it and I can't remember why.