ros-industrial-attic/ur_modern_driver #223: Using /joint_speed

BigHopes, 2018-10-30:
Hello all,

I am trying to do a speed control in a UR5 robot and using python.
Please would you give me an example of publishing a message, I think the starting commands are as follows, but couldn't complete.

`pub=rospy.Publisher('/joint_speed', trajectory_msgs.msg.JointTrajectory,queue_size=30)`
`pub.publish("speedj([ ???   ])")` 

Regards,

gavanderhoorn, 2018-11-01:
In principle with the `joint_speed` topic you need to publish a `JointTrajectory` with a single `JointTrajectoryPoint` in it. Populate the `velocities` field with the desired velocities.

The [jog_arm](https://github.com/UTNuclearRoboticsPublic/jog_arm) package does this. You could take a look at it.

gavanderhoorn, 2018-11-01:
The recently added support for `JointGroupVelocityControllers` might be a nicer way to do this though.

gavanderhoorn, 2018-11-01:
As this is not really an issue with the code in this package I'm going to close the issue.

Feel free to keep commenting on it though.

closed 2018-11-01 by gavanderhoorn