ros-industrial-attic/ur_modern_driver #231: UR with force sensor

BigHopes, 2018-11-07:
Hi all, 

I am working with UR5, CB3 and want to read the force sensor values (optoforce) in the same (python) file that read the robot values (position..). 

I downloaded the force sensor driver and built it as explained in  ([src](https://github.com/OptoForce/etherdaq_ros)) and I can run the demo file (roslaunch optoforce_etherdaq_driver demo.launch) which display the force values on the terminal.

Can I read the force values within the file that read the robot states?
I have the ur_moder_driver in a different workspace than the force, is it a problem? 
Also, I found that the force sensor (Optoforce) node publish a topic "/etherdaq_data_raw" in Wrench or in WrenchStamped format, I tried to write a code to read it but didn't work.

Please can you help me, I think it is not a difficult issue but I tried a lot and didn't work. 
Thanks,

gavanderhoorn, 2018-11-07:
I'm not sure what you're asking but I have a feeling that it's not actually something to do with `ur_modern_driver` per se.

Subscribing to two different topics in the same node is definitely possible. There is no difference in writing two subscribers for `std_msgs/String` or `geometry_msgs/WrenchStamped`. Then use the timestamps in the header of the `sensor_msgs/JointState` messages and the `WrenchStamped` to sync the two datastreams.

gavanderhoorn, 2018-11-07:
As this is not an issue with the code in this repository but a more general support request I would suggest to post this as a question on ROS Answers.

closed 2018-11-07 by gavanderhoorn